mavlink/qgroundcontrol #7225: Maximum number Vehicles

tdard, 2019-02-13:
## Current Behavior
QGroundControl cannot handle more than 8 Vehicles

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Download QGroundControl source code
2. Build & Run the project from Qt Creator
3. Create 8 PX4 mock links
4. The 8th mock link makes the application crash.

I also tried using 7 mock links & 1 true Pixhawk (PX4), the application still crashes.
I am working on a project involving drone swarm (15-30 links), and therefore I worry a bit..

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 16.04 LTS
- QGC Version: Master branch from 31 Jan.
- QGC build: self-built from source

Should I had downloaded a stable version such as the one we can find in the branch Stable_V3.4 ?

Thank you

DonLakeFlyer, 2019-02-15:
The max number of vehicles supported by QGC is currently 7. Also the current code is not very helpful when you exceed this max it just stops connecting. I'll look at making this better in 3.6

tdard, 2019-02-16:
Thank you sir,

Though, I need to increase the number of vehicles so I can have a real swarm for my project.
Could you tell me:
-What file(s) should be modified in order to increase the number of vehicles ?
-What do I risk if I try to increase the number of Vehicles ? 

I am not afraid of going deep into the code, so I don't ask you any favor more than this.

Best regards

DonLakeFlyer, 2019-02-16:
I don't know the answer to that yet. This is internal to mavlink and how it uses MAVLINK_COMM_NUM_BUFFERS and internal data structures. I believe there is a mechanism to allow this to be dynamic as opposed to a fixed number. But I'm not sure yet.

tdard, 2019-02-16:
Thank you sir, 

Once I success in increasing the number of vehicles, I let you know. 

Best regards

tdard, 2019-02-17:
Initial value of  MAVLINK_COMM_NUM_BUFFERS is defined to 16 in the mavlink library, in the file mavlink_types.h
I modified its value statically to 32 and I could connect 14 vehicles.
To do that, I undefined this macro and re-defined it in the concerned files of QGroundControl. I got all these files using:
`grep -nr "MAVLINK_COMM_NUM_BUFFERS"`   
The result is shown here:

`libs/mavlink/include/mavlink/v2.0/mavlink_helpers.h:29:	static mavlink_status_t m_mavlink_status[MAVLINK_COMM_NUM_BUFFERS];

libs/mavlink/include/mavlink/v2.0/mavlink_helpers.h:46:	static mavlink_message_t m_mavlink_buffer[MAVLINK_COMM_NUM_BUFFERS];

libs/mavlink/include/mavlink/v2.0/mavlink_types.h:172: * applications can set MAVLINK_COMM_NUM_BUFFERS to the maximum number

libs/mavlink/include/mavlink/v2.0/mavlink_types.h:176:#ifndef MAVLINK_COMM_NUM_BUFFERS

libs/mavlink/include/mavlink/v2.0/mavlink_types.h:178:# define MAVLINK_COMM_NUM_BUFFERS 16

libs/mavlink/include/mavlink/v2.0/mavlink_types.h:180:# define MAVLINK_COMM_NUM_BUFFERS 4

src/comm/QGCMAVLink.h:23:extern mavlink_status_t m_mavlink_status[MAVLINK_COMM_NUM_BUFFERS];

src/comm/MAVLinkProtocol.h:106:    uint64_t    totalReceiveCounter[MAVLINK_COMM_NUM_BUFFERS];  ///< The total number of successfully received messages

src/comm/MAVLinkProtocol.h:107:    uint64_t    totalLossCounter[MAVLINK_COMM_NUM_BUFFERS];     ///< Total messages lost during transmission.

src/comm/MAVLinkProtocol.h:108:    float       runningLossPercent[MAVLINK_COMM_NUM_BUFFERS];   ///< Loss rate

src/QGCApplication.cc:130:mavlink_status_t m_mavlink_status[MAVLINK_COMM_NUM_BUFFERS];
`
 /!\  I added the modification only in src files

By now, it is enough to solve my problem. 

Thank you a lot for your help,

Best regards

Jaeyoung-Lim, 2019-12-15:
I was able to simulate 10 vehicles in SITL gazebo with QGC v3.5.5

Video: https://youtu.be/Mskx_WxzeCk

What would be the limit of the number of vehicles?

waltersajtos, 2021-09-21:
Any updates on this issue? Any pointers would be awesome!

Jaeyoung-Lim, 2021-09-21:
@waltersajtos Do you have any problems? This should be working now

Jaeyoung-Lim, 2021-09-21:
@waltersajtos http://docs.px4.io/master/en/simulation/multi_vehicle_simulation_gazebo.html 

This will get you to around ~250 vehicles. I will start there

liang-hong, 2025-06-04:
What is the max number of vehicles in QGC V4.4.3 and later？ A swarm with over 60 vehicles is in need for me to show in a map. 
The swarm includes quadrotors， VTOLs， UGVs and soon. the SYS_IDs are settings on qgc connecting with pixhawk(fmu_v6x version) from 1 to 60. The mavlink data used by qgc(on the ground computer) is routered by mavlink_router package on Jetson Orin NXs(on vehicles) in the local area network. For example, the IP of ground computer is 192.138.10.100, then the udp prot of mavlink from vehicles with SYS_ID 1&2 are respectively 192.168.10.100:15501, 192.168.10.100:15502.

DonLakeFlyer, 2025-06-06:
The information in the above thread hasn't changed. Read through it, modify QGC and you'll have to test yourself.

DonLakeFlyer, 2025-07-16:
No idea whether it will work or not. As said above it's all about cpu usage and memory usage. All variable on tablet cpu, tablet memory, vehicle telemetry, maybe even vehicle connection type. I doubt there is a one size fits all table.

antonymarion, 2025-07-16:
> No idea whether it will work or not. As said above it's all about cpu usage and memory usage. All variable on tablet cpu, tablet memory, vehicle telemetry, maybe even vehicle connection type. I doubt there is a one size fits all table.

Ok we will do the test anyway 😄 and discover the sizing required in terms of both cpu and memory usage

Jaeyoung-Lim, 2025-07-16:
@antonymarion I could simulate over 200 vehicles without much issue on QGC side.

You are limited with the SYS_ID field in mavlink, which is only up to 255 minus broadcast and QGC itself.