mavlink/qgroundcontrol #7249: Plan is waiting on terrain data endlessly

nsteele, 2019-02-25:
## Expected Behavior
Successful upload of missions where the waypoints' altitudes are specified "Above Terrain"

## Current Behavior
When clicking "Upload Required", I see "Unable to Save/Upload ... Plan is waiting on terrain data from server for correct altitude values"

## Steps to Reproduce:

1. Open QGroundControl with "UDP" as chosen in the "AutoConnect to the following devices" section of the "General" application settings.
2. In a shell, navigate to the px4 firmware workspace and run the following
```
export PX4_HOME_LAT=40.44977
export PX4_HOME_LON=-79.8999
make px4_sitl_default jmavsim
```

3. Once QGroundControl connects to the simulation, plan a waypoint mission where at least one of the waypoints' altitudes is specified as "Above Terrain"

4. Click "Upload Required"

## System Information
- Operating System: Ubuntu 18.04
- QGC Version: v3.5.5
- QGC build: stable
- Flight Controller: px4 jmavsim simulation
- Autopilot (with version): PX4 1.8.0

## Detailed Description

## Log Files and Screenshots

Here is an example waypoint with altitude specified "Above Terrain"
[image]

Here is the error message
[image]

Here is the QGC console log. It keeps printing the same thing.

[image]

DonLakeFlyer, 2019-02-25:
Looks like some sort of AirMap problem...

nsteele, 2019-02-25:
Is there anyway to confirm this is an AirMap problem? I'd be satisfied closing this if that were the case. I'm wondering if the QGC url or token or something is incorrect.

As a workaround for me, is there a facility in QGC for using a different terrain map server?

DonLakeFlyer, 2019-02-25:
> Is there anyway to confirm this is an AirMap problem?

Working on it. Don't close the issue. You can't use a different terrain server.

DonLakeFlyer, 2019-02-25:
AirMap is figure out how to fix it.

kikislater, 2019-02-26:
Agree with @nsteele : Airmap is crap and does not help drone industry... Making money with non requiring authorisation !!! 
Why choosing this platform ? And Airmap uses SRTM90 which is not accurate. SRTM30 is now available.

DonLakeFlyer, 2019-02-26:
@kikislater You are certainly welcome to submit a pull request to QGC to add support for any terrain sever you would like.

kikislater, 2019-02-26:
Or forking the project

nsteele, 2019-02-26:
Do I need to set an API key and Client ID in the AirMap section of the Application Settings to enable waypoints with altitudes relative to terrain?

[image]

dogmaphobic, 2019-02-26:
> Do I need to set an API key and Client ID in the AirMap section of the Application Settings to enable waypoints with altitudes relative to terrain?

No, that's for something else. The elevation API didn't require any authentication. Airmap shutdown that (open) API and we are waiting for them to tell us how to access it.

DonLakeFlyer, 2019-02-27:
Unless something goes horribly wrong AirMap will turn the support back on. There was a misunderstanding as to what api we were using for what in QGC. Hopfully won't take too long.

Antiheavy, 2019-02-27:
I tested this 2 days ago and elevation download was not working.  I just tested now and elevation download is working again.

DonLakeFlyer, 2019-02-27:
It's all working again now.

closed 2019-02-27 by DonLakeFlyer