mavlink/qgroundcontrol #7252: Joystick is not detected in v3.5.0 mac release

patrickelectric, 2019-02-26:
## Expected Behavior
Detect the joystick.

## Current Behavior
Do not detected joystick.

## Steps to Reproduce:
1. Connect joystick
2. Open QGC
3. Device is not detected

## System Information
- Operating System: 10.14.2
- QGC Version: 3.5.0

[image]

[log.txt](https://github.com/mavlink/qgroundcontrol/files/2907167/log.txt)

DonLakeFlyer, 2019-02-26:
QGC uses SDL lib for joysticks. So if SDL doesn't see it QGC won't see it. Only possible fix is is newer version of SDL can see it.

DonLakeFlyer, 2019-02-26:
Or is this a new problem where it was working before?

patrickelectric, 2019-02-26:
Yes, it was working before, but since I did some updates in the OS, that may be the cause.