mavlink/qgroundcontrol #7253: Camera parameter miss in v1.9.0

abdulmajid92, 2019-02-27:
Hi.
Anyone have same problem with me?
In px4 firmware v1.8.2, the camera paramater showm like below
[image]

but after I upload px4 v1.9.0dev, i miss the camera paramater (also in paramters tab).
[image]

DonLakeFlyer, 2019-02-27:
@dogmaphobic The Camera page is visible based on the existence of the TRIG_MODE parameter. Did something change in the firmware such that this param no longer comes down?

dogmaphobic, 2019-02-28:
I have not dealt with that since I added the code many years ago. That would be a question to the firmware folks.

closed 2026-10-05 by github-actions[bot]