mavlink/qgroundcontrol #7254: Mission transfer failed

Polarbeargo, 2019-02-27:
## Expected Behavior
Hello everyone, I'm trying to upload mission plan to Intel aero both on Yocto and Ubuntu.

## Current Behavior
Both OS have the same problem when i upload mission as following screen shots:
[image]

## System Information
[image]

## Detailed Description
As I upload the mission plan got "Mission transfer failed. Retry transfer. Error: Vehicle didn't request all items from ground station: Mission request. Can anyone give me advice to resolve this mission transfer failed?

DonLakeFlyer, 2019-02-27:
This looks like a really noisy comm link. You can see loss rates in Settings/Mavlink page. You can turn on logging for mission upload/download: https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html to see exactly which is happening. Doubtful it's a QGC side problem so closing. I'll reopen if the problem can be pointed at QGC.

DonLakeFlyer, 2019-02-27:
@hamishwillee This is another candidate for a troubleshooting section in the docs. Happens a lot.

closed 2019-02-27 by DonLakeFlyer

hamishwillee, 2019-02-28:
@DonLakeFlyer You already did that: https://docs.qgroundcontrol.com/en/Support/PlanUploadDownload.html :-)

The fact that this user did not find that tells me that we need to make it more visible. So what I have done is moved this into the Plan page (cross linked from common problems). I also added the actual message text - so this information would show up in search. 

Please review: https://github.com/mavlink/qgc-user-guide/pull/200

Polarbeargo, 2019-02-28:
Thank you @DonLakeFlyer , now I'm able to see what happen during mission transfer fail 👍 , like start MAVLink log command fail, no internet access, network not available. My RTSP Video streaming and telemetry data is working great and i can remote ssh to Intel aero. Anyone can help me about how to trouble shooting further to solve mission transfer fail?
[image]
[image]

hamishwillee, 2019-02-28:
@Polarbeargo What is the loss rate on this dialog: https://docs.qgroundcontrol.com/en/SettingsView/MAVLink.html#link-status

Polarbeargo, 2019-02-28:
Thank you @hamishwillee, I got the log with the document instruction :) but as I try to find link status QGroundControl appear like the screen shots not sure what I am missing.
[mission.txt](https://github.com/mavlink/qgroundcontrol/files/2913081/mission.txt)
[image]

dogmaphobic, 2019-02-28:
It looks like you are running a very old version of QGC. It's missing a great deal of features added to that panel in the past several months:

[image]

Polarbeargo, 2019-02-28:
@dogmaphobic Thank you :)! I will also upgrade my QGroundControl.

hamishwillee, 2019-02-28:
QGC 3.5 (current) has the section.

Polarbeargo, 2019-02-28:
Here is my  loss rate and newest log.
[mission transfer fail.txt](https://github.com/mavlink/qgroundcontrol/files/2913186/mission.transfer.fail.txt)
[image]

hamishwillee, 2019-02-28:
There you are then. As it says in the docs, if you are in the high single digits (ie around 9) you will get intermittent failures, and too much higher the chance of fail gets to 100%.  Why is the environment you are in so noisy? Can you get the vehicle closer to the ground station before uploading a mission?
You might need a better antenna or radio. The software can't really help you in this case - the messages just aren't getting through.

Polarbeargo, 2019-02-28:
@hamishwillee, The Drone is near me about 5 cm, I'll try to test in another environment very appreciate your guiding, thanks 👍 !

dogmaphobic, 2019-02-28:
That's an awful lot of messages (~180k) How long it took to get that many messages? If it was a short time, the data rate may be saturating the link.