mavlink/qgroundcontrol #7261: Purple Cube (Slim/Mini) Not autoconnecting with 3.5.0 or latest Daily builds

kklemens, 2019-03-01:
## Expected Behavior
Purple Cube should autoconnect to QGC without having to manually connect through the connection manager.

## Current Behavior
Purple Cube is not autoconnecting to v3.5.0 or latest Daily builds of QGC. Can only connect by going into Comm Links -> Add -> Serial -> Connect.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Loaded latest version of ArduRover (v3.5.0) onto Purple Cube from Mission Planner (unable to load from QGC)
2. Plugged USB cable directly into computer and Purple Cube.
3. Unable to connect.

## System Information

- Operating System: Windows 10
- QGC Version: v3.5.0 or latest Daily build
- QGC build: Stable, Daily
- Flight Controller: Purple Cube (Pixhawk 2.1 Mini/Slim)
- Autopilot (with version): ArduRover 3.5.0, Copter latest beta, Plane, latest beta

## Detailed Description
N/A

## Log Files and Screenshots
https://discuss.cubepilot.org/t/autoconnect-issue-with-cube-mini-purple/181/9

DonLakeFlyer, 2019-03-22:
Looks like there are new Windows USB Drivers which need to be installed. New Mission Planner beta has them I beleieve. QGC does not yet have them. I'll fix that. 

If you have those already installed can you turn on logging for LinkManagerVerboseLog (https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html). And then attach the output for connecting to the board here.

DonLakeFlyer, 2019-03-23:
Nope, this is just busted. Hopefully fixed in pull.

DonLakeFlyer, 2019-03-27:
@kklemens Can you try tomorrow's daily buils and let me know how it works.

kklemens, 2019-03-27:
Will do! I'll get you a report of everything I run through.

kklemens, 2019-03-29:
@DonLakeFlyer Tried the most recent build for today and no luck. No autoconnect with a pre-loaded ChibiOS firmware and unable to load a new firmware. Screenshot attached when I tried. 
[image]
Also, log from the console: 
[Console Log with Slim Cube 3-29-19.txt](https://github.com/mavlink/qgroundcontrol/files/3024148/Console.Log.with.Slim.Cube.3-29-19.txt)

DonLakeFlyer, 2019-03-30:
The screenshot is from flashing the bootloader not firmware flashing. That error means that the firmware you are running doesn't support the command to flash the bootloader.

DonLakeFlyer, 2019-03-30:
@tridge How come this board is not showing up with correct USB information:
```
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:523 - "portName:           "COM7""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:524 - "systemLocation:     "\\\\.\\COM7""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:525 - "description:        "USB Serial Device""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:526 - "manufacturer:       "Microsoft""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:527 - "serialNumber:       "270054000751383034343237""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:528 - "vendorIdentifier:   11694"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:529 - "productIdentifier:  4117"
```

DonLakeFlyer, 2019-03-30:
I string match on Manufacturer as a fallback but it is not showing up as "Hex/ProfiCNC"

DonLakeFlyer, 2019-04-05:
@kklemens  Can you try tomorrow's daily build and see how it works?

kklemens, 2019-04-07:
Thanks Don! I'm out of the office for the next week, but I'll give it a try when I get back.

kklemens, 2019-04-15:
@DonLakeFlyer Connection Success! It worked with the latest Plane v3.9.7. I wiped the board and installed the firmware from QGC. It seems to work with the only error being "Check BRD_TYPE: Unable to detect board". In checking the BRD_TYPE it sets the parameter to AUTO.

I also installed the latest MP .msi so that may have had the drivers too.

DonLakeFlyer, 2019-04-16:
@kklemens Thanks for letting me know. I'll take a look at that BRD_TYPE thing.

DonLakeFlyer, 2019-04-19:
@kklemens When you flash the Purple Cube, in the status text for the flashing in the ui. Can you tell me what firmware file it is lading? It should be listed there. Want to make sure it flashed the Purple specific firmware.

kklemens, 2019-04-19:
[image]

DonLakeFlyer, 2019-04-19:
Nope. That is the generic FMU V3 firmware as opposed to the specific Purple firmware.

mjisted, 2019-08-20:
Just bought a batch of Purple Cubes - unable to install PX4 firmware on nearly all of them.  Here's where I am at:
- Installed Windows 10 drivers to recognise cube as ProfiCNC Cube F4 FMU
- Initially installed Purple Cube firmware (Ardupliot) using Mission Planner - this is necessary as the Cube does not connect to QGC.
- At this stage, autoconnect works with QGC and connection possible with MP.
- Using MP, installed downloaded PX4 (1.9.2).  
- Neither QGC or MP connect.  MP Freezes on connection and QGC never connects.  QGC warns of device not responding.

I have one Purple Cube that does work - QGC reports firmware as 'Unknown' (bit it's in HTL mode).