mavlink/qgroundcontrol #7263: Feature request - Crop sprayer (400$ award)

DangerD256, 2019-03-04:
1). Add new survey grid for crop sprayer with options:
- Fly altitude
- Spacing (same as on survey grid)
- Angle (same as on survey grid)
- Fly speed
Turnaround distance should be zero and not editable (to avoid going outside field, to avoid crash to the trees)
2) Add ability to add obstacles to grid (Like on DJI agras software). 
[image]
[image]

In addition:
3) Missions manager - save/load missions and add sync function to synchronize them with remote server. (list of field names with groups on left)

hamishwillee, 2019-03-04:
@DonLakeFlyer being able to add an exclusion within a survey looks like it might be useful. The only other way to reasonably do this would be to use (dynamic) obstacle avoidance in the flight stack.

DangerD256, 2019-03-05:
Can you ask him?) its a bit urgent for us as season starts and we need to spray fields

jstrebel, 2019-03-05:
It would be very useful if we could import or draw **several** static objects which the UAV has to fly "around"

hamishwillee, 2019-03-05:
@DangerD256 I wouldn't be reliant on Don or anyone to add this feature in any particular timeframe. In the meantime you could inefficiently "fake" this kind of thing using a mission that has multiple surveys bracketing the areas you wish to omit.

DangerD256, 2019-03-05:
10 hectares field with approximately 10 trees... Drawing a lot of separate surveys is a pain.. And if I will make "bracketing" it will also don't work

GlidLov, 2019-03-09:
+1 on the adding-obstacle planning

DonLakeFlyer, 2019-03-22:
If I implemented this I would support it from GeoFence. Or in other words you set up GeoFence exclusion for the trees or what not and then the Survey generation would respect those exclusion areas. This way there isn't some new thing needed to make out an exclusion for say a tree.

Note: For now this would also mean it would only work on PX4 Pro since ArduPilot doesn't support current MAVLink GeoFence spec.

@antiheavy I'd assume this would be helpful for you as well?

Antiheavy, 2019-03-22:
This feature request has a lot of value for multicopters and ground/surface vehicles.  It has some value for fixed wings, but much less.

I thought breaking areas into multiple survey's was going to be the approach to this?  Maybe that proved impractical?

Geofence is another reasonable way to approach this and would probably work well for multicopter and ground/surface vehicles.  Safety buffers may need to be built in to avoid accidentally breaching the Geofence and triggering possible failsafe actions.  It also makes it even less useful for fixed wing, but there isn't much you can do about that.

An important compliment to this is improved triggering at the edges of survey areas.  This is an open PR:  https://github.com/mavlink/qgroundcontrol/pull/5532

chris89116, 2020-07-06:
I am also interested in this type of option can you tell me if you have already advanced on the subject ?
Thank you

hamishwillee, 2020-07-06:
Well, ArduPilot have now added Geofence support. Nothing else AFAIK. 
It would be cool if a survey automatically planned around geofence areas.

palucki, 2021-12-07:
Has anyone advanced on this?
I thought about following the proposal from: 
[Coverage Path Planning: The Boustrophedon Cellular Decomposition](https://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.69.2482&rep=rep1&type=pdf)
But it seems the model used for survey generation is far away from being compatible with it.

Also I think this issue and [6969](https://github.com/mavlink/qgroundcontrol/issues/6969) are duplicates?

Drdroneman, 2022-10-13:
Hello Developers, 
     Is there any development on Crop sprayer features in Qgroundcontrol android version? or heading hold (i.e. Conditional YAW) in QGC?

khoacc, 2023-05-07:
Please make it available, I add more 400$

flavves, 2026-10-15:
I'd like to investigate and work on this issue

DangerD1024, 2026-10-15:
> 18\. I'd like to investigate and work on this issue

Not needed at the moment as I have damn war in my country...(

flavves, 2026-10-15:
> 19\. > 18. I'd like to investigate and work on this issue
>      
>      
>      Not needed at the moment as I have damn war in my country...(

I am so sorry to hear about your situation. Please take care of yourself and stay safe—that is what truly matters right now.

In the meantime, I went ahead and worked on this request so it’s ready whenever you or anyone else needs it. If it can be of any help, I’ll be very glad.

I’ve submitted the PR for obstacle-aware survey planning:

https://github.com/mavlink/qgroundcontrol/pull/15301

Briefly, it handles polygon/circular GeoFence exclusions with clearance buffers, optimizes route segments (reduced test route by ~24%), and passes local survey tests.

Wishing you and your loved ones peace and safety.