mavlink/qgroundcontrol #7265: This is a bug to not find xxx.px4 file.

roseyangyu, 2019-03-05:
This is a bug.I have download QGC-v3.4.0 and v3.4.0+.When I upload firmware by QGC,I can't find the that Firmware download from github.It's hidden.I can input its name to upload in QGC.But version 3.3 is different,I can find FW in QGC-v3.3

roseyangyu, 2019-03-05:
[image]
[image]
This is V3.5.0 and in same folder.
3.3 is normal.

DonLakeFlyer, 2019-03-23:
Hmm, this doesn't reproduce for me. I wonder if it only happens in the translated version.

bys1123, 2019-03-23:
@DonLakeFlyer I think maybe translation team use the wrong  brackets "（）"，I updated it.

DonLakeFlyer, 2019-03-26:
Ah, that would make sense. @dogmaphobic How do translated builds flwo through the system. Will a new daily just pick this up?

dogmaphobic, 2019-03-26:
> Will a new daily just pick this up?

No, the process is quite manual. Once the translation is fixed, it must be "built" (within Crowdin), downloaded, compiled (.ts to .qm) and a new PR made with the changes.

DonLakeFlyer, 2019-12-24:
Can anyone verified this is working with latest daily build translations?

bys1123, 2019-12-25:
@DonLakeFlyer This is already fixed long time ago on stable release.

DonLakeFlyer, 2019-12-25:
Thanks for letting me know

closed 2019-12-25 by DonLakeFlyer