mavlink/qgroundcontrol #7266: QGC shows weird values with mavlink2 messages

patrickelectric, 2019-03-05:
Mavproxy output:
```
< SERVO_OUTPUT_RAW {time_usec : 610825080, port : 0, servo1_raw : 1500, servo2_raw : 1500, servo3_raw : 1500, servo4_raw : 1500, servo5_raw : 1500, servo6_raw : 1500, servo7_raw : 0, servo8_raw : 1500, servo9_raw : 1100, servo10_raw : 0, servo11_raw : 0, servo12_raw : 0, servo13_raw : 0, servo14_raw : 0, servo15_raw : 0, servo16_raw : 0}
```
QGC output:
[image]

DonLakeFlyer, 2019-12-11:
@patrickelectric Can you check this again. It seems fine in latest daily

DonLakeFlyer, 2019-12-24:
Closing unless it's verified to be bad again

closed 2019-12-24 by DonLakeFlyer