mavlink/qgroundcontrol #7268: Linux Daily Build (2019-03-06) won't start

jbeyerstedt, 2019-03-06:
## Expected Behavior
QGC starts. It works with the build from 2019-02-05 (HEAD:e44b51403).

## Current Behavior
When starting QGC with double click, nothing happens. When starting from a terminal the following error message is shown:
```
execv error: : No such file or directory
```

## Steps to Reproduce:
1. Download daily build appimage from the User Guide
2. make it executable (`chmod +x`)
3. execute (./QGroundControl.AppImage)
4. QGC will not start

## System Information
When posting bug reports, include the following information
- Operating System: ubuntu 18.04 LTS/ ubuntu 18.04 LTS
- QGC build: daily 2019-03-06

stefandunca, 2019-03-06:
#7269 will fix this

dogmaphobic, 2019-03-06:
#7269 is merged. Please confirm the fix and we can close this.

jbeyerstedt, 2019-03-08:
The latest Daily Build works fine, so this issue is resolved.

closed 2019-03-08 by jbeyerstedt