mavlink/qgroundcontrol #7275: Joystick inputs does not work after motor test

patrickelectric, 2019-03-06:
After doing motor test and arming the vehicle, it's not possible to send joystick inputs.
After disarm the vehicle, a "motor test failed" message appears.

jaxxzer, 2019-03-20:
This sounds like it is possibly a problem with the firmware, or a cooldown imposed by the firmware. I've tried things here, and everything is working as expected. I was not able to reproduce this.

You need to be more specific about the sequence of events to reproduce this. For example, the usual workflow is to configure motor directions, then arm with joystick and drive the vehicle around with the joystick. Given that, it's suspicious that this is the only report of such an issue.

jaxxzer, 2019-03-20:
Do any of the messages in this file appear under the vehicle messages? https://github.com/ArduPilot/ardupilot/blob/master/ArduSub/motors.cpp

jaxxzer, 2019-04-05:
This problem is actually expected behavior when you remain on the motor test page. If you are on the motor test page, motor test commands are sent to the autopilot when it is armed. When the autopilot gets motor test commands, it ignores joystick input.

There is another bug involved, and that is that the code/timers in the vehicle setup qml pages keeps running after the fly view is selected. So if motor test page is open _in the background_ (while fly view is active), then it is sending motor test commands after you arm the vehicle. I think there may need to be a new issue opened for that.

closed 2019-07-17 by jaxxzer