mavlink/qgroundcontrol #7277: QGC behaves as a spam from the 90s in mac after opening a log file

patrickelectric, 2019-03-07:
## Expected Behavior
Behaves as a normal window.

## Current Behavior
ITS FREAKING JUMPING AROUND.

## Steps to Reproduce:
1. Open QGC 3.5.0 dmg file in mac
2. Open a log file
3. Have fun.

## System Information
- macOS 10.14.3
- QGC Version: 3.5.0
- QGC build: release
- Flight Controller: log file of ArduSub stable

[image]

[2019-03-06 18-27-10.tlog.zip](https://github.com/mavlink/qgroundcontrol/files/2942354/2019-03-06.18-27-10.tlog.zip)

DonLakeFlyer, 2019-03-22:
Wow, that's an interesting one!

DonLakeFlyer, 2019-12-11:
This is a different in daily builds. Working fine now.

closed 2019-12-11 by DonLakeFlyer