mavlink/qgroundcontrol #7278: Error on link Serial Device on ttyUSB0. Error connecting: Could not create port. Permission error while locking the device

gian1312, 2019-03-08:
I get the following error while connecting the telemetry connection.

error on link Serial Device on ttyUSB0. Error connecting: Could not create port. Permission error while locking the device

I tried the following steps to get rid of the issue
Add to dialout
remove modemanager
Untik everything (not UDP) in the general options

`student@ubuntu:~/Desktop$ ls -l /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 Mar  8 10:49 /dev/ttyUSB0
student@ubuntu:~/Desktop$ groups
student adm dialout cdrom sudo dip plugdev lpadmin sambashare
`
I rebooted the system. Still the same issue

## System Information
When posting bug reports, include the following information
- Ubuntu 16.04
- QGC 3.5.0
- QGC build: [stable]
- Flight Controller: [e.g., Pixhawk I,]
- Autopilot (with version): [e.g., PX4 1.8.2]

hamishwillee, 2019-03-12:
This might be relevant https://github.com/mavlink/qgroundcontrol/issues/6666

hamishwillee, 2019-03-12:
@DonLakeFlyer @dogmaphobic  FYI, I'm planning on adding a connecting to QGC page (following on from https://github.com/mavlink/qgroundcontrol/issues/7189), so I'll add a note about this issue too.

gian1312, 2019-03-12:
I have already done that. "Solution" is to compile and flash the firmware over Linux and use QGC for windows to use telemetry.

DonLakeFlyer, 2019-03-22:
@hamishwillee I don't understand what the issue/solution is?

hamishwillee, 2019-03-25:
@DonLakeFlyer I think this is a duplicate of https://github.com/mavlink/qgroundcontrol/issues/6666

What @gian1312 is saying is that they couldn't resolve it, so they just started using Windows instead of Linux.

What I was saying is that I could add some troubleshooting text as described in https://github.com/mavlink/qgroundcontrol/issues/6666

gian1312, 2019-03-25:
As I stated in the issue description and in my response, the unticking thing was the first thing I did. I could not resolve this issue on Linux. I am using the Windows version now.

DonLakeFlyer, 2019-03-26:
#6666 is about connecting to a device with a non-default baud rate. Make sense to doc that somewhere. This issue seems like just regular connection problem. I don't think you get permisison denied for baud rate problems.

hamishwillee, 2019-03-27:
@DonLakeFlyer While you are undoubtably right, the other case did report permission error: 

> "Error on link Test1. Error connecting: Could not create port. Permission error while locking the device".

Anyway, leaving this with you to debug.

arahimfahim40, 2020-02-12:
Hey guys, I don't have the exact error but another error is prompted which I can't find any solution to, any help will be greatly appreciated.

Using Navio 2 with usb adapter telemetry; connecting it prompts the error: "Error on link Serial Device COM3. Error connecting: Could not create port. Access is denied."

Similar error prompts when trying to connect to Mission Planner.

0b10010010, 2020-06-02:
Hello,

I am working with navio2 with px4 and I get the same "permission error while locking the device". Is it possible to work with navio2 without WiFi telemetry? I am only asking since 3DR SiK radio will give us much longer mission distance away from QGC.

Thanks in advance,

AK

0b10010010, 2020-06-03:
Hello @arahimfahim40,

I don't know if you are still wondering about establishing the connection between QGC and Navio2 using telemetry radio but I was able to mitigate this issue by deleting the lock file on the host computer (running QGC) and starting the mavlink on Navio2 with ttyAMA0 (serial UART telemetry radio for my case). Please let me know if you need the details.

arahimfahim40, 2020-06-04:
Hello @0b10010010 , thank you for the reply and concern. I actually solved my problem and it is described here: https://community.emlid.com/t/uart-port-broken-what-should-i-do/17142/60

heeplr, 2024-02-10:
Got the same Issue with a fresh build of current master (no config, no cache):

```Open failed "Permission error while locking the device" QSerialPort::PermissionError "rover" autconnect: true```

While system permissions are fine, I launch qgroundcontrol with strace and found that even without configured comm link, there is frequent access to the /dev/ttyUSB0 my telemetry adapter is using (about once or twice per second):

```
readlinkat(AT_FDCWD, "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/ttyUSB0/subsystem", "../../../../../../../bus/usb-ser"..., 4096) = 35
openat(46, "ttyUSB0", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 42
openat(46, "ttyUSB0", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 42
openat(AT_FDCWD, "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/ttyUSB0/tty/ttyUSB0/uevent", O_RDONLY|O_NOCTTY|O_CLOEXEC) = 31
read(31, "MAJOR=188\nMINOR=0\nDEVNAME=ttyUSB"..., 4104) = 34
openat(46, "ttyUSB0", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 42
openat(46, "ttyUSB0", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 42
```

I tried to quickly press the "connect" button after plugging in the adapter in hope to win the race, which works. Although the link seems slowish.

In my case, it's definetly qgroundcontrol trying to open the configured ttyUSB0 multiple times, while only the first time can succeed.

Is there some sort of device scan active in QGroundcontrol/Qt ?

( I'm using QtSerialPort-6.6.1 )

DonLakeFlyer, 2024-02-10:
Qt tries to autoconnect to devices on USB: Radios, Autopilot boards, RTK GPS an so forth. You can turn off autoconnect from settings. It in the dropdown in the toolbar for "click to manually connect":
[image]

heeplr, 2024-02-10:
> Qt tries to autoconnect to devices on USB: Radios, Autopilot boards, RTK GPS an so forth. 

Should it, though? Even when there's no comm link configured and the user didn't request any sort of "autodetection" in the configure phase?

> You can turn off autoconnect from settings. It in the dropdown in the toolbar for "click to manually connect": 

Thank you, that fixed it.

DonLakeFlyer, 2024-02-11:
I'm going to close this since it looks like the original problem is really old

closed 2024-02-11 by DonLakeFlyer