mavlink/qgroundcontrol #7279: firmware probleme

Philouzik, 2019-03-10:
Hello,
I am a new in flight controller user.
I want to learn how to use Qgroundcontrol with a Px4 2.4.8.
Maybe, I made a mistake because I buy a pack readytosky and I have a problem for the setting of the firmware.
Can be compatibility problem?
I launch QGC, I go into the settings and firmware, so I connect the px4 usb but no choice appears?
This PX4 is not recognized?

USB port works well
The sd is installed, blank and goes well through the same usb port!
I use a mac on Sierra 10.13.4
I hope to be in the right place for these issues.
thank you for your clarification.

hamishwillee, 2019-03-11:
@Philouzik The terminology of your question is confusing. Pixhawk is flight controller hardware. PX4 is flight controller software/firmware. So I think you are saying that you have a Pixhawk 2.4.8 flight controller and you want to connect it to QGC to update the firwmare. Is that correct?

I suggest you ask on the discussion boards: https://docs.qgroundcontrol.com/en/Support/Support.html#support  because even if it doesn't work, this will not be a bug with QGC.

DonLakeFlyer, 2019-03-22:
This should go to discussion boards.

closed 2019-03-22 by DonLakeFlyer