mavlink/qgroundcontrol #7282: gst_parse_launch error due to Protocol buffer version mismatch

RohanKurane, 2019-03-11:
Hi,

This issue is related to Gstreamer and not Mavlink or QGC. 
I apologize for opening a bug here, but I cannot seem to open a bug in Gstreamer/gstreamer. If anyone can move it there, that would be great. 

My problem is that when I run the function gst_parse_launch (pipeline, &error) from my application, I get a error for protocol buffers. 
`[libprotobuf FATAL google/protobuf/stubs/common.cc:79] This program was compiled against version 2.6.1 of the Protocol Buffer runtime library, which is not compatible with the installed version (3.6.1).  Contact the program author for an update.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "/build/mir-O8_xaj/mir-0.26.3+16.04.20170605/obj-x86_64-linux-gnu/src/protobuf/mir_protobuf.pb.cc".)`

I cant imagine gstreamer being dependent on protocol buffers, but is that the case here ? 

I am running ubuntu 16.04.

RohanKurane, 2019-03-11:
Is there an alternative for get_parse_launch() ? I see all it does it call gst_parse_launch_full() who calls priv_gst_parse_launch_full().

DonLakeFlyer, 2019-03-22:
No idea where the gstreamer forums/issue place is. But nobody here is likely to be able to help.

closed 2019-03-22 by DonLakeFlyer