mavlink/qgroundcontrol #7286: Add telemetry overlay to saved video

jaxxzer, 2019-03-12:
This is one of the most highly requested features of sub users. The idea is to have telemetry such as depth, location, time etc overlaid on the video file in order to produce and maintain better records and deliverables.

There are subtitle formats ([ssa](https://en.wikipedia.org/wiki/SubStation_Alpha)) that allow positioning and formatting of the text. Media players such as VLC have support for rendering these subtitle files. QGC could produce these subtitle files with the telemetry data synchronized to the video. In order to keep it simple, we recommend limiting the available telemetry overlay to a set of <=20 or so common values like gps coordinates, altitude, temperature etc. One thought was to just use the available 'vehicle values widget' (pictured below) and add a column like 'export' to choose to add a value to the subtitle file. We would like to force the text placement also, for example a title and date at the top right, and all telemetry values in the bottom left (no user configuration).

[image]

We have considered several other approaches, but we favored the subtitle file for these reasons:
- The original video remains available without any 'burned in' overlay
- Media players already support rendering the subtitle files and burning them into the video file in post, we don't have to implement any of the rendering side ourselves
- As minimal as possible user configuration - easy for them!
- Implementation requires simply producing the subtitle file which is in ascii format - easy for us!

If you guys approve of the approach, @patrickelectric @Williangalvani and I can do the work to make it happen.

hamishwillee, 2019-03-12:
FWIW I love the idea. @dogmaphobic ?

dogmaphobic, 2019-03-12:
We've had this conversation in the past. I had done some research. I even still have the code. Adding it is quite simple. The issue was that you have no control over the layout of the text. The protocol is meant for subtitles, which means sentences at the bottom of the screen. It works for a few bits of data but nothing like the image above.

patrickelectric, 2019-03-12:
@dogmaphobic 	we did some study around the available formats, SSA allows control of the text position and layout.

dogmaphobic, 2019-03-12:
Ha! I have the code that loads the telemetry logs and parses each message. From there, you pick whatever you want and write it out to the subtitle file along with the time stamp. Let me see if I can find it and I can start the PR. You can pick it up from there.

Williangalvani, 2019-03-14:
This is an example of what we can have using the SubStation Alpha format:
[image]

[This](https://gist.github.com/Williangalvani/e01dabb2f860c6831781eb4016f00b45) is the subtitle file used. You can use any of [these](https://en.wikipedia.org/wiki/SubStation_Alpha#Players_and_renderers) players to test it, Just use a random video and load this subtitles file.

DonLakeFlyer, 2019-03-22:
Would be nice if this could be done in combination with a rework of the Values Widget UI. Why not make it work for both somehow?

jaxxzer, 2019-03-22:
What else would you be interested in addressing regarding the values widget @DonLakeFlyer?

jaxxzer, 2019-04-04:
Ping @DonLakeFlyer for details relating to reworking the values widget as we go at this.

DonLakeFlyer, 2019-04-05:
Basically make this setup be a replacement for the values widget. Always have the values show up directly on the screen whether it be video or map.

jaxxzer, 2019-04-09:
@DonLakeFlyer I think the function and implementation of the qml widgets vs the subtitle file may actually need to be quite separate.

I'm wondering if your underlying interest is in making making the values widgets less obstructive than they are today, or something else. Is there an open issue regarding this?

DonLakeFlyer, 2019-04-19:
The point is to get rid of the values widget and instead allow the user to put the values overlayed directly on the screen. Whether that be map or video. Would need to be thought throug from a real estate standpoint though since the value page thing would still be there.

jaxxzer, 2019-04-19:
Ok Don, I think that patch will be independent of this work, but may be done in conjunction.

Also @Williangalvani is closing in on a working branch for the feature described here: 
[image]

closed 2019-06-03 by dogmaphobic

LPNautic, 2019-11-11:
Hi,
When can we have this function in Ardusub Stable ??
Thanks

Williangalvani, 2019-11-12:
Hello, 
This was merged into QGC, but as QGC is going through a lot of changes right now  this isn't on any stable releases yet. You can try it with the [daily builds](https://docs.qgroundcontrol.com/en/releases/daily_builds.html) meanwhile.

PR #7482 (2019-06-03): Implement overlay on recorded videos by using subtitle files
This patch allows QGC to record a data overlay while recording the video. This is implemented by producing an independent subtitle file for use with each video that is recorded to file.  The subtitle file is a text file in the [SubStation Alpha](https://en.wikipedia.org/wiki/SubStation_Alpha) format. This is supported by your usual media players, where you can control if the overlay is displayed on playback, or optionally combine the video and subtitle file and export a single video file with the subtitles 'burned in'.

This patch wi [... cut]