mavlink/qgroundcontrol #7289: Parameters missing

twistedwings, 2019-03-15:
Alot of parameters are missing in qgroundcontrol.
They are in missionplanner.
E.g. RCXX_OPTION: RC input option

qgroundcontrol 3.5
arducopter-heli 3.7.0-dev

giacomo892, 2021-06-23:
@DonLakeFlyer Regarding this issue, in general, would be possible to add an option to allow the user to suppress the "missing parameters" error? It's quite annoying.

DonLakeFlyer, 2021-06-24:
@giacomo892  That is a real error. Either QGC it out of date with the firmware you are using or the reverse. It indicates some part of QGC ui will fail. Is this specific error still happening with latest QGC. If so, add details and I'll fix.

giacomo892, 2021-06-24:
When you for instance disable some code in ardupilot then the parameters of that part will be missing and QGC will complain. That's it basically

DonLakeFlyer, 2021-06-24:
That isn't something that QGC is setup to run against. QGC is built to run against standard released versions of ArduPilot. If you want a version of QGC that works against a custom version of Ardupilot you also need your own custom modified version of QGC which has had the features which correspond to be removed from QGC as well.

closed 2021-06-24 by DonLakeFlyer

giacomo892, 2021-06-24:
Well not really, Ardupilot is cutting a lot of features on F4 boards and the messages are a bit annoying.

giacomo892, 2021-06-29:
Flash Plane 4.1beta official release on an F4 board and:
[image]

Also bare in mind that custom build server is coming soon to ardupilot, so sooner or later this should be addressed. At least having an option to disable the warnings.