mavlink/qgroundcontrol #7296: QGC 3.5.0 OSX: cannot select custom firmware files for update

bluefix, 2019-03-19:
- Operating System: [macOS Mojave 10.14.1]
- QGC Version: [3.4.5 stable]
- Flight Controller: [Pixhawk 4 mini]
- Autopilot: [PX4 1.8.0 beta]

while trying to update the firmware of a Pixhawk 4 mini:

Vehicle Setup
       - Firmware 
          --  Advanced settings (selected)
              --- Custom firmware file...

in the opening Finder-window it is not possible to select a firmware file .PX3 or .BIN. 
All files names are grey, only directories are selectable.

This procedure is possible in QGC 3.4.4

DonLakeFlyer, 2019-03-22:
I'm not quite sure which version you are reporting this against? Title says 3.5.0, details say 3.4.5. I checked latest Stable 3.5.1 and latest daily and it was working fine for me.

Also, there is no such thing as a .px3 file. They are .px4 files.

closed 2019-03-23 by bluefix

DonLakeFlyer, 2019-03-23:
Is this working correctly for you now?

bluefix, 2019-03-24:
my error: 
my report is about version 3.5.0 (MAC) and i was looking for .px4 files. 
In ver. 3.5.0 no custom firmware  files could be selected. 
The ver. 2.4.5 was OK and the daily  (032219) is OK again. 
BTW the default firmware update 1.8.2 (stable) does not work on my Pixhawk 4 mini: IO error, no radio, no blue LEDs, 2nd magn. sensor error. Version 1.8.0beta is OK. But that should be no fault of QGC.

Thank you for patience, Jo

Joachim Buehrmann-Schulz 
Hardyweg 8
14055 Berlin, Germany
Phone +4930 2131758
Mobile +49163 4783294
jo@3jo.de   jobs@gmx.de

DonLakeFlyer, 2019-03-26:
Are you using  German translated version? There was a problem with Chinese version which was a bad translation of the file dialog selection extensions which broke Chinese. Wondering if German may have been the same.

> BTW the default firmware update 1.8.2 (stable) does not work on my Pixhawk 4 mini: IO error, no radio, no blue LEDs, 2nd magn. sensor error. Version 1.8.0beta is OK. But that should be no fault of QGC.

You should report that to firmware folks. QGC doesn't have control over that.

dogmaphobic, 2019-03-26:
> Wondering if German may have been the same.

This was fixed here: https://github.com/mavlink/qgroundcontrol/pull/7250