mavlink/qgroundcontrol #7299: Ardupilot can't reset parameters to defaults

jaxxzer, 2019-03-20:
QGC 3.5.0 on ubuntu 16.04, ArduSub 3.5.3

There is no longer any option to reset the parameters to the vehicle defaults. It used to be present in this menu:

[image]

DonLakeFlyer, 2019-03-22:
That's because it's not support by ArduPilot.

jaxxzer, 2019-12-04:
> That's because it's not supported by ArduPilot.

How do you figure? It certainly was working in the past.

DonLakeFlyer, 2019-12-10:
I don't think ArduPilot supports the mavlink way of doing this:

```
    _vehicle->sendMavCommand(MAV_COMP_ID_ALL,
                             MAV_CMD_PREFLIGHT_STORAGE,
                             true,  // showError
                             2,     // Reset params to default
                             -1);   // Don't do anything with mission storage
```

jaxxzer, 2019-12-17:
https://github.com/ArduPilot/ardupilot/blob/f8f591c7db6def816421588c97b0e32ef4901a52/libraries/GCS_MAVLink/GCS_Common.cpp#L3689

I can follow this up with a pr after confirming things.

DonLakeFlyer, 2019-12-24:
I turned this on. Hope it doesn't break anything!

PR #8173 (2019-12-25): Fixes
* Fix #6508. Add new Fly View: Keep Map Centered On Vehicle option
* Fix #7299. Allow reset to default for any firmware type
closed 2019-12-24 by DonLakeFlyer