mavlink/qgroundcontrol #7301: QGC says vehicle did not respond to request for parameters,can not display full interface when connect to pixhawk via telemetry

crossa, 2019-03-21:
QGC  v3.5.0
OS    macOS 10.14.3  android 6.0.1
Pixhawk 2.4.8 with px4 1.8.2

crossa, 2019-03-21:
If qgc connect to pixhawk via usb cable, it's fine and qgc can got all of the parameter from pixhawk.

julianoes, 2019-03-21:
@crossa what telemetry thing do you use?

crossa, 2019-03-21:
This telemetry is  a replica of 3dr radio telemetry. In fact, it's just a wireless serial port. 
I have tried the lastest arduplot and lastest APM planner with the same radio telemetry ,I found it's fine.

The issue appeared after QGC@3.2.3.

@julianoes

julianoes, 2019-03-21:
@crossa that's good information but I don't know what the issue could be.

crossa, 2019-03-21:
I have read the dev guide of QGC . It says QGC requires QT@5.11.0 and above on macOS but I still remember that the early version of QGC required QT@5.5 . Do you know whether the QGC package for macOS contained the correct redistributable package of QT or not?  Maybe it's a issue of Qt.
Do you guys konw the maintainer of QGC?
Please @ him 

@julianoes @dogmaphobic @LorenzMeier

crossa, 2019-03-21:
@julianoes @dogmaphobic @LorenzMeier
I found if  both 'Emit heatbeat' and  'Request'  start of MAVLink telemetry streams' checked
the loss rete grower up to 90%

crossa, 2019-03-21:
@julianoes @dogmaphobic @LorenzMeier 
I think this is  a bug  of the firmware. I rollback my QGC to 3.1.3. The issue can not be resolved.
So I try to flash the follwing firmware into my pixhwak

1.  px4 1.8.2
2. the lastest stable version of nuttx-arducopter  for pixhawk
3. ArduCopter 3.6.7

The results are clear.

If  using px4 1.8.2 and nuttx-arducopter, both QGC and APM planner can't get all the parameter via radio telemetry.

When using nuttx-arducopter , APM planner still can't get all the parameter via radio telemetry

But the  ArduCopter without nuttx  works fine !

Please check the firmware.  The issue may be a mistake of the configuration of the firmware may also
be a issue of nuttx.  I don't know how the debug pixhawk and there's no record of the problem in the log. Please check the source code and configuration of px4 firmware

DonLakeFlyer, 2019-03-22:
If you are getting high loss rates, you will never get parameter sync to complete. Nor should you ever try to fly anything in this state. QGC doesn't have any control over this. This would be a hardware setup problem.

closed 2019-03-22 by DonLakeFlyer

crossa, 2019-03-23:
I have tried anthor firmware. If using arducopter without nuttx ,qgc can get all of the parameter . I have reported the issue to the px4 team.