mavlink/qgroundcontrol #7310: Feature: Add declination setup

patrickelectric, 2019-03-22:
Moved from: bluerobotics/qgroundcontrol#143
Original author: @jaxxzer

### Issue description
The autopilot has a declination lookup, but it won't use it without gps lock. We should have manual declination entry on sensors/compass setup.

jaxxzer, 2020-12-10:
This should be a part of compass calibration step, selecting auto declination or manual declination + entry. The compass calibration will be affected if the declination currently used by the autopilot is not valid (ie if there is no gps, or gps does not have a lock).

jaxxzer, 2020-12-10:
@Williangalvani

jaxxzer, 2020-12-16:
@DonLakeFlyer Would it be acceptable to add a declination setup to the compass sensor calibration step?

In the right sidebar that appears we could have a choice of auto declination/manual declination, and if the user selects manual declination, provide a manual entry field. This is performed before the compass calibration sequence initiates.

DonLakeFlyer, 2020-12-16:
My concern here is exposing something to all users for which 99% of them will never use and may just cause confusion. If I have it wrong in that most people will never use this, then fine. But if I have it right and maybe this is mainly a sub thing (whereas almost always all other vehicles have a gps) then I would bracket it for showing up only for subs.

PR #9307 (2021-01-11): Create field for manual Magnetic Declination setup for Submarines
This introduces a Magnetic Declination field for Subs:
[image]

Two possible discussion points:

 - ~Should we show it as degrees instead of radians ? Is something already there in the fact system that could help with the conversion?~
 - Having the checkbox inverted (user clicks "Manual declination" instead of "Automatic Declination") makes more sense from a user perspective. This could be done by adding an "inverted" property to FactCheckBox, but [this binding](https://github.com/mavlink/qgroundcontrol/blob/master/src/FactSyste [... cut]
closed 2021-01-11 by DonLakeFlyer