mavlink/qgroundcontrol #7320: Windows: Install new px4 drivers

DonLakeFlyer, 2019-03-23:
The px4driver.msi in QGC is out of date. And it doesn't look like a new version is available for use as a standalone installer. May need to change to DPINST*.exe installation mechanism with Driver directory from Mission Planner instead. Still checking.

DonLakeFlyer, 2019-03-23:
Related to #7261

RichLewis007, 2019-04-18:
Where is the source code for px4driver, and what does it do? I see it being installed when installing QGC in Windows, but I can't find the source or explanation of what this installs anywhere. Thanks!

DonLakeFlyer, 2019-04-18:
I'm not really sure. The source comes from Michael Oborne the Mission Planner dev. The files are in the MP github repo.

RichLewis007, 2019-04-29:
Thanks for the reply. As an Open Source project, shouldn't the location and contents of the px4driver software be more transparent? Maybe forked to a Mavlink owned repo? As it is, when we install QGroundControl the px4driver software is installed on our system without our permission, explanation, or indication of where it comes from and what it does.

DonLakeFlyer, 2019-04-29:
> As it is, when we install QGroundControl the px4driver software is installed on our system without our permission, explanation, or indication of where it comes from and what it does.

Huh? The drivers have there own install UI which says what they are and you can cancel if you like.

dogmaphobic, 2019-04-29:
It's an USB->Serial (COM Port) "driver". This is present in all platforms but Windows, hence the need to install the "driver".

RichLewis007, 2019-04-30:
Thanks for the info! In trying to find the source for this executable that Windows users typically install during the installation of QGC, px4driver.msi, I searched the MissionPlanner repo:

https://github.com/ArduPilot/MissionPlanner/search?q=px4driver&unscoped_q=px4driver

I searched the other ArduPilot and PX4 repos for this source code, without results.

I guess you got the latest version of the executable from here, and put it in the mavlink/qgroundcontrol repo? http://firmware.ardupilot.org/Tools/MissionPlanner/

This seems insecure, as the verification checksums in the checksums.txt file there aren't themselves signed by a known trusted source. So, an attacker would only need to replace the px3driver.msi file there, edit the checksums.txt file, and all of a sudden all of our good QGC and MissionPlanner Windows users would end up installing Malware. Am I missing something, or is this something that should be addressed?

Without any access to the source code for this binary blob we are distributing far and wide when installing this open source flight controller, there's no way to check it for safety.

What did you mean by "May need to change to DPINST*.exe installation mechanism"?

Thanks for your help! I know you were just trying to get the latest and best version of this driver for the community. I just want to be sure we're safe.

DonLakeFlyer, 2019-04-30:
Your going to to have ask the Mission Planner folks those question since they produce this.

DonLakeFlyer, 2019-12-11:
Drivers install has been updated

PR #7384 (2019-04-27): New Windows drivers
Fix #7320
closed 2019-12-11 by DonLakeFlyer