mavlink/qgroundcontrol #7321: qrc:qml/FirmwareUpgrade.qml:116: TypeError: Type error

troffel1, 2019-03-25:
Hi there,

Pixhawk V5 only connects once to QGroundcontrol (with serial connection). It successfully connected one time, after firmware upgrade (V.1.8.2). 
The second time that the board is connected, QGroundControl states: "Waiting for vehicle connection".
In the "Firmware" tab, instructions state to start firmware upgrade (again). When the same firmware is 'upgraded', no re-connection is made, and clicking "Summary" tab refers that the vehicle should be upgraded, creating a loop. 

This happens both on QGC v3.5.0 stable and the daily build.
The FirmwareUpgradeLog can be found below.

My OS is linux 16.04.

Does somebody know what's going on? Thanks!

qrc:qml/FirmwareUpgrade.qml:116: TypeError: Type error
Filter rules "*Log.debug=false\nFirmwareUpgradeLog.debug=true\n"
FirmwareUpgradeLog: PX4FirmwareUpgradeThreadController::findBoard
FirmwareUpgradeLog: 0 "PX4 FMU V5"
qrc:qml/FirmwareUpgrade.qml:116: TypeError: Type error
FirmwareUpgradeLog: Found px4 stable version "v1.8.2 Stable Release"
FirmwareUpgradeLog: Found px4 beta version "v1.7.4 Beta Release"
FirmwareUpgradeLog: Board gone
FirmwareUpgradeLog: _findBootloader "/dev/ttyACM0"
FirmwareUpgradeLog: 0 "PX4 FMU V5"
FirmwareUpgradeLog: Found bootloader
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Copter/stable/PX4-heli/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Copter/stable/fmuv5/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Sub/stable/fmuv5/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Copter/beta/PX4-heli/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Plane/latest/PX4/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Copter/beta/fmuv5/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Sub/beta/fmuv5/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Copter/latest/fmuv5-heli/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Plane/stable/PX4/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Copter/stable/fmuv5-heli/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Plane/beta/PX4/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Rover/latest/PX4/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Copter/beta/fmuv5-heli/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Plane/latest/fmuv5/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Rover/stable/PX4/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Plane/stable/fmuv5/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Rover/beta/PX4/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Copter/latest/PX4/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Sub/latest/PX4/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Plane/beta/fmuv5/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Rover/latest/fmuv5/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Copter/stable/PX4/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Sub/stable/PX4/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Rover/stable/fmuv5/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Copter/beta/PX4/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Copter/latest/PX4-heli/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Sub/beta/PX4/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Copter/latest/fmuv5/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Rover/beta/fmuv5/git-version.txt"
FirmwareUpgradeLog: Downloading "http://firmware.ardupilot.org/Sub/latest/fmuv5/git-version.txt"
FirmwareUpgradeLog: Download complete "http://firmware.ardupilot.org/Plane/stable/fmuv5/git-version.txt" "/tmp/git-version.txt"
FirmwareUpgradeLog: Adding version to map, version:firwmareType:vehicleType "ArduPlane V3.9.6" FirmwareUpgradeController::FirmwareType_t(StableFirmware) FirmwareUpgradeController::FirmwareVehicleType_t(PlaneChibiOSFirmware)
FirmwareUpgradeLog: Download complete "http://firmware.ardupilot.org/Copter/beta/fmuv5/git-version.txt" "/tmp/git-version.txt"
FirmwareUpgradeLog: Adding version to map, version:firwmareType:vehicleType "ArduCopter V3.6.7" FirmwareUpgradeController::FirmwareType_t(BetaFirmware) FirmwareUpgradeController::FirmwareVehicleType_t(CopterChibiOSFirmware)
FirmwareUpgradeLog: Download complete "http://firmware.ardupilot.org/Plane/stable/PX4/git-version.txt" "/tmp/git-version.txt"
FirmwareUpgradeLog: Adding version to map, version:firwmareType:vehicleType "ArduPlane V3.9.6" FirmwareUpgradeController::FirmwareType_t(StableFirmware) FirmwareUpgradeController::FirmwareVehicleType_t(PlaneFirmware)
FirmwareUpgradeLog: Download complete "http://firmware.ardupilot.org/Copter/beta/PX4-heli/git-version.txt" "/tmp/git-version.txt"
FirmwareUpgradeLog: Adding version to map, version:firwmareType:vehicleType "ArduCopter V3.6.7" FirmwareUpgradeController::FirmwareType_t(BetaFirmware) FirmwareUpgradeController::FirmwareVehicleType_t(HeliFirmware)
FirmwareUpgradeLog: Download complete "http://firmware.ardupilot.org/Sub/latest/fmuv5/git-version.txt" "/tmp/git-version.txt"
FirmwareUpgradeLog: Adding version to map, version:firwmareType:vehicleType "ArduSub V3.6.0-dev" FirmwareUpgradeController::FirmwareType_t(DeveloperFirmware) FirmwareUpgradeController::FirmwareVehicleType_t(SubChibiOSFirmware)
FirmwareUpgradeLog: Download complete "http://firmware.ardupilot.org/Copter/latest/PX4-heli/git-version.txt" "/tmp/git-version.txt"
FirmwareUpgradeLog: Adding version to map, version:firwmareType:vehicleType "ArduCopter V3.7.0-dev" FirmwareUpgradeController::FirmwareType_t(DeveloperFirmware) FirmwareUpgradeController::FirmwareVehicleType_t(HeliFirmware)
FirmwareUpgradeLog: Download complete "http://firmware.ardupilot.org/Copter/beta/fmuv5-heli/git-version.txt" "/tmp/git-version.txt"
FirmwareUpgradeLog: Adding version to map, version:firwmareType:vehicleType "ArduCopter V3.6.7" FirmwareUpgradeController::FirmwareType_t(BetaFirmware) FirmwareUpgradeController::FirmwareVehicleType_t(HeliChibiOSFirmware)
FirmwareUpgradeLog: Download complete "http://firmware.ardupilot.org/Plane/beta/PX4/git-version.txt" "/tmp/git-version.txt"
FirmwareUpgradeLog: Adding version to map, version:firwmareType:vehicleType "ArduPlane V3.9.6" FirmwareUpgradeController::FirmwareType_t(BetaFirmware) FirmwareUpgradeController::FirmwareVehicleType_t(PlaneFirmware)
[... 14 lines cut ...]
FirmwareUpgradeLog: Download complete "http://firmware.ardupilot.org/Copter/latest/fmuv5-heli/git-version.txt" "/tmp/git-version.txt"
FirmwareUpgradeLog: Adding version to map, version:firwmareType:vehicleType "ArduCopter V3.7.0-dev" FirmwareUpgradeController::FirmwareType_t(DeveloperFirmware) FirmwareUpgradeController::FirmwareVehicleType_t(HeliChibiOSFirmware)
FirmwareUpgradeLog: Download complete "http://firmware.ardupilot.org/Rover/stable/PX4/git-version.txt" "/tmp/git-version.txt"
FirmwareUpgradeLog: Adding version to map, version:firwmareType:vehicleType "ArduRover V3.4.2" FirmwareUpgradeController::FirmwareType_t(StableFirmware) FirmwareUpgradeController::FirmwareVehicleType_t(RoverFirmware)
FirmwareUpgradeLog: Download complete "http://firmware.ardupilot.org/Plane/beta/fmuv5/git-version.txt" "/tmp/git-version.txt"
FirmwareUpgradeLog: Adding version to map, version:firwmareType:vehicleType "ArduPlane V3.9.6" FirmwareUpgradeController::FirmwareType_t(BetaFirmware) FirmwareUpgradeController::FirmwareVehicleType_t(PlaneChibiOSFirmware)
FirmwareUpgradeLog: Download complete "http://firmware.ardupilot.org/Rover/latest/fmuv5/git-version.txt" "/tmp/git-version.txt"
FirmwareUpgradeLog: Adding version to map, version:firwmareType:vehicleType "ArduRover V3.6.0-dev" FirmwareUpgradeController::FirmwareType_t(DeveloperFirmware) FirmwareUpgradeController::FirmwareVehicleType_t(RoverChibiOSFirmware)
[... cut ...] Firmware)
FirmwareUpgradeLog: _flash stackType:firmwareType:vehicleType FirmwareUpgradeController::AutoPilotStackType_t(AutoPilotStackPX4) FirmwareUpgradeController::FirmwareType_t(StableFirmware) FirmwareUpgradeController::FirmwareVehicleType_t(DefaultVehicleFirmware)
FirmwareUpgradeLog: PX4FirmwareUpgradeThreadWorker::_flash
FirmwareUpgradeLog: PX4FirmwareUpgradeThreadWorker::_erase
FirmwareUpgradeLog: Erase complete
FirmwareUpgradeLog: Program complete
FirmwareUpgradeLog: Verify complete
Done

sritank, 2020-01-21:
I'm having the same issue. Firmware upgrade is successful but QGC doesn't reconnect and I don't get the setup for Airframe, radio, etc. Did you solve this?

sritank, 2020-01-21:
I found that the daily build below did not have this issue. Hope this is fixed in the future.

https://docs.qgroundcontrol.com/en/releases/daily_builds.html

closed 2026-10-05 by github-actions[bot]