mavlink/qgroundcontrol #7322: Export layman's log (.csv output!)

jaxxzer, 2019-03-26:
In a world of mavlink, missions, binary logs, xmls, and proxies, working out how to produce a log with the information one is after can be a daunting and time consuming task, even for advanced users. At the time of writing, Blue Robotics' best recommendation for producing a .csv from a log file requires a python installation and command line tools. This is far from easy as it can be.

This is a proposal for a simple-as-possible solution that will serve a majority of users and use cases.

Most users will already have some idea of how they would like to handle the data they are interested in. They usually want to import data into a spreadsheet program or other analysis tool like matlab. These tools support importing .csv files. 

The proposed solution is for QGC to export a .csv file with some ~20-50  or so curated fields/columns such as gps coordinates, altitude, attitude, temperature, etc. To keep things simple, the keys chosen should be human readable and self explanatory (including units!). Whatever values/keys we decide to log, there won't be any user configuration; all columns/keys that we choose will always be logged. Spreadsheet programs will be able to filter/render the data however is required in post. The decision to forgo user configuration means that all available data will be in the original csv (no forgetting to turn on logging) while making the implementation very simple (no ui, no configuration), and maintaining minimal overhead.

One suggestion for deciding the columns/keys to be logged is to just use everything that appears as an option for display in the vehicle values widget, shown below. This is the same suggestion as put forward in the [video overlay](https://github.com/mavlink/qgroundcontrol/issues/7286) proposal.

[image]

jaxxzer, 2019-04-03:
We are eager to work a PR together, but we don't want to sink any time into something that will not be accepted. @DonLakeFlyer @dogmaphobic what do you think of the proposal?

DonLakeFlyer, 2019-04-03:
To me this is best left as a standalone tool. We'll never get to the point where it makes people happy without adding tons of code:
* What fields to output
* In what order
* What format
* On and on and on

I'm keeping QGC away from log file analysis type tasks and leaving those for standalone tools. Otherwise QGC just bloats out to this big massive thing with a huge pile of confusing options to wade through.

jaxxzer, 2019-05-28:
Hi Don. I understand the considerations you put forward, and that is exactly why the proposal is worded the way that it is, and the suggestion is to keep the implementation as **simple as possible**. There are **no user configurable options**. QGC runs and a .csv appears just like the .tlog, no user intervention necessary, minimal overhead and code.

The idea here is that this simple 'something' is infinitely more useful than the 'nothing' that we currently have.

PR #7477 (2019-09-04): Add csv logging of telemetry
This makes each vehicle log it's facts to a `.csv` file named as `<timestamp> vehicle<vehicle_ID>.csv`
This is a sample file:  [2019-05-21 18-22-17_vehicle2.txt](https://github.com/mavlink/qgroundcontrol/files/3228873/2019-05-21.18-22-17_vehicle2.txt)

This requires **no interaction of the user whatsoever**, the .csv will be created automatically just like the .tlogs.

This allows users to easily open the data with any spreadsheet software or Matlab, without going through the hassle of using mavlogdump or Mission Planner to deal with a tlog, for example.

close #73 [... cut]
closed 2019-09-04 by DonLakeFlyer