mavlink/qgroundcontrol #7324: Camera capture status isn't handled correctly for photos

tpwrules, 2019-03-26:
When a capture starts, the camera manager sets its status to in progress, and resets it to idle when it receives a CAMERA_CAPTURE_STATUS message which says the camera is now idle. But when you take a photo or photo sequence, the manager never starts the timer to send a MAV_CMD_REQUEST_CAMERA_CAPTURE_STATUS.

Here, it resets the request count as if it is about to start the timer:
https://github.com/mavlink/qgroundcontrol/blob/6af7e8f2f40f390bb17f3b56e6c910010165787e/src/Camera/QGCCameraControl.cc#L364-L365

And here, it starts the timer to re-request the status if the start capture command failed:
https://github.com/mavlink/qgroundcontrol/blob/6af7e8f2f40f390bb17f3b56e6c910010165787e/src/Camera/QGCCameraControl.cc#L681-L688

But it never starts the timer when the command succeeds, like it does here for video:
https://github.com/mavlink/qgroundcontrol/blob/6af7e8f2f40f390bb17f3b56e6c910010165787e/src/Camera/QGCCameraControl.cc#L656-L663

I'm not sure what the correct solution is and I'm not sure what the obligation is for the camera to send CAMERA_CAPTURE_STATUS periodically. However, if my camera sends it right after it sends the CAMERA_IMAGE_CAPTURED message even if it never receives a request, everything works correctly.

dogmaphobic, 2019-03-27:
The camera is supposed to send a `CAMERA_IMAGE_CAPTURED` message for each image it captures. QGC will not request it.

Video is a bit different. When QGC sends a `MAV_CMD_VIDEO_START_CAPTURE` command, it sets param2 to 0, which means it is telling the camera not to send `CAMERA_CAPTURE_STATUS` automatically when recording video. QGC will request it on its own periodically. The reason for this is historic as this point as the camera that caused issues (it wasn't respecting param2) has since been fixed.

tpwrules, 2019-03-27:
When QGC takes a photo, it sets the `_photoStatus` to `PHOTO_CAPTURE_IN_PROGRESS`:
https://github.com/mavlink/qgroundcontrol/blob/6af7e8f2f40f390bb17f3b56e6c910010165787e/src/Camera/QGCCameraControl.cc#L364

My camera sends `CAMERA_IMAGE_CAPTURED` when it captures an image, and this causes the total image counter to increment and an image marker to show up on the map. 

But QGC still believes the image capture is in progress, so clicking the button to capture another image does not send another `MAV_CMD_IMAGE_START_CAPTURE`. The way to restore this functionality seems to be to send a `CAMERA_CAPTURE_STATUS` with `image_status` equal to 0, i.e. idle. This causes QGC to set the image status back to idle here:
https://github.com/mavlink/qgroundcontrol/blob/6af7e8f2f40f390bb17f3b56e6c910010165787e/src/Camera/QGCCameraControl.cc#L1460-L1463

which then allows this test to pass:
https://github.com/mavlink/qgroundcontrol/blob/6af7e8f2f40f390bb17f3b56e6c910010165787e/src/Camera/QGCCameraControl.cc#L352-L354

and another photo to be taken. QGC might need to send a `MAV_CMD_REQUEST_CAMERA_CAPTURE_STATUS` so that the photo status gets set back to idle, or not set it to in progress. Or maybe the onus is on my camera to automatically send a `CAMERA_CAPTURE_STATUS` when the capture status changes. I'm not sure what the proper behavior is based on reading the spec here: https://mavlink.io/en/services/camera.html .

dogmaphobic, 2019-03-27:
Hmm... I see it. By the time my camera (firmware) sends its `CAMERA_CAPTURE_STATUS`, it's all done with the capture and the status is already idle. That's why I never run into this before.

The idea behind having the camera send the `CAMERA_CAPTURE_STATUS` on its own is primarily due to the way missions work. During a mission, QGC has no idea when the camera is taking pictures so the camera must tell it when it does. However, when using the camera interactively, QGC does these requests to find out when the camera is idle again.

QGC is indeed not starting the timer and that, along with the fact you sent a non idle status brought this up. The code even handles a continuous non idle state:

https://github.com/mavlink/qgroundcontrol/blob/6af7e8f2f40f390bb17f3b56e6c910010165787e/src/Camera/QGCCameraControl.cc#L1467-L1470

But that won't happen until it starts the timer when it first sends the `MAV_CMD_IMAGE_START_CAPTURE` command.

I will add that to https://github.com/mavlink/qgroundcontrol/pull/7325

tpwrules, 2019-03-27:
I'm still not sure if my camera is doing the correct order of operations. Here's what happens from my camera's perspective:

1. Camera receives `MAV_CMD_IMAGE_START_CAPTURE` with param2=0, param3=1
2. Camera takes picture. This is a fast process for now.
3. Camera sends `CAMERA_IMAGE_CAPTURED` with time and GPS position etc.
4. To work around this issue in QGC, camera sends `CAMERA_CAPTURE_STATUS` with `image_status=0`. 

My camera will never send a `CAMERA_CAPTURE_STATUS` unless it receives a `MAV_CMD_REQUEST_CAPTURE_STATUS`, except in step four above. It will also never sends a `CAMERA_CAPTURE_STATUS` with `image_status != 0`.

So is step four required in the protocol? Should my camera send `CAMERA_CAPTURE_STATUS` messages when the capture status changes, or is it only supposed to send them when asked by the GCS?

dogmaphobic, 2019-03-27:
You are right. It’s been over two years since I wrote this and even I got confused. I indeed mixed `CAMERA_IMAGE_CAPTURED` with `CAMERA_CAPTURE_STATUS`. What you are doing is exactly what you should be doing. Step 5 above was only needed as a work around for the issue in question. Once the fix is merged, you will no longer need to do that.

Let’s start over from scratch, while disregarding all of the above:

@hamishwillee, this probably needs some work and documented accordingly. His implementation was correct. The problem came up due to a bug in QGC. Nonetheless, the amount of digging required to figure it out could be avoided if we properly document this sequence.

The camera receives a MAV_CMD_IMAGE_START_CAPTURE. It does what it needs to do and ACKs the command like any other command (with `MAV_RESULT_ACCEPTED` or otherwise)

Whenever an image is captured, the camera will send a `CAMERA_IMAGE_CAPTURED` message with information about the image it just captured.

The above happens whether the command came from the ground station (user interactively taking pictures) or from the autopilot (missions). This message is sent to component ID = 0 as both the autopilot and the ground station might want this message.

When the user is interactively taking pictures from the ground station, the ground station will send `MAV_CMD_REQUEST_CAPTURE_STATUS` commands to find out the camera’s current status. It will continue sending this command until the camera is again _Idle_.

I’m sure this is still clear as mud but it’s a start.

@tpwrules. https://github.com/mavlink/qgroundcontrol/pull/7325 is merged and you should be able to test with it. Thanks for bringing this up.

hamishwillee, 2019-03-28:
@dogmaphobic 

> This message is sent to component ID = 0 as both the autopilot and the ground station might want this message.

Interestingly, since CAMERA_IMAGE_CAPTURED has no target system/component ID you can't actually send this specifically to anyone - it is by default broadcast. There is always a sender component/system id, and that should usually be the camera. Ie it is a little odd to use the sender id on the message as filter for interst.

Anyway, for missions ..>
What I think you are saying is that the sequence for missions is as below - is this correct?
[image]

For user interactively taking pictures I THINK what you are implying is that the user can only take individual images. 
- Is this correct?
- In this case you check if camera is OK to use, then allow them to send message (if it is, otherwise they can't actually press the button?) then, check if idle again. So something like this?:

[image]

For both images, please confirm that's right/where the errors are, then I can embed them in the docs with some further explanation

dogmaphobic, 2019-03-28:
> Interestingly, since CAMERA_IMAGE_CAPTURED has no target system/component ID

Again writing without thinking. I was confusing messages with commands.

> What I think you are saying is that the sequence for missions is as below - is this correct?

That's exactly it! Thanks.

hamishwillee, 2019-03-28:
@dogmaphobic Great. Please review the PR - https://github.com/mavlink/mavlink-devguide/pull/176

tpwrules, 2019-03-31:
Yes, this seems to work now.

dogmaphobic, 2019-03-31:
Closing as it seems it's been all fixed.

kuiskas, 2020-02-09:
Hi,
sorry to reopen this thread, but I am curious how did it all end I was just testing the camera protocol and still getting this same problem @tpwrules described.
Running PX4 1.10.1 and QgroundControl 3.5.6 on a Pixhawk cube black and a pixhawk 4.
The Process I am following is the one described in mavlink camera protocol
MAV_CMD_REQUEST_CAMERA_INFORMATION      --> Ack     --> mavlink_msg_camera_information_pack
MAV_CMD_REQUEST_CAMERA_SETTINGS         --> Ack     --> mavlink_msg_camera_settings_pack 
MAV_CMD_REQUEST_STORAGE_INFORMATION     --> Ack     --> mavlink_msg_storage_information_pack
MAV_CMD_REQUEST_CAMERA_CAPTURE_STATUS   --> Ack     --> mavlink_msg_camera_capture_status_pack

Qgrouncontrol build the instrument panel for the camera. Once pressed on the camera trigger button:
MAV_CMD_IMAGE_START_CAPTURE             --> Ack     --> mavlink_msg_camera_image_captured_pack

This only is sent once and does allow to take more pictures. 
I also tried QGroundControl daily build in order to see if it was something new for that.
I am also getting at aprox 0.5Hz a MAV_CMD_DO_CONTROL_VIDEO and I do not know what PX4 or QGroundcontrol are expecting. 
Hope someone can help me with this issue.
T

dogmaphobic, 2020-02-11:
> Running PX4 1.10.1 and QgroundControl 3.5.6

Does this happen with daily builds?

kuiskas, 2020-02-11:
Yes, I tried too. Maybe I am sending something wrong, but I've been looking long time and playing a lot with the params, same behavior.

kuiskas, 2020-02-27:
Hi, 
I tried the new version 4.0.0 of QGC referent to this topic, it is better but from time to time after a couple of pictures it stills freezes. The only way to get it back from the this frozen mode is to send a camera_capture_status messages as @tpwrules said. 
Again the procedure is the following:
1- long MAV_CMD_REQUEST_CAMERA_INFORMATION
2- long MAV_CMD_REQUEST_CAMERA_SETTINGS
3- long  MAV_CMD_REQUEST_STORAGE_INFORMATION
4- long MAV_CMD_REQUEST_CAMERA_CAPTURE_STATUS
then trigger through QGC:
5- long  MAV_CMD_IMAGE_START_CAPTURE
6- long MAV_CMD_REQUEST_CAMERA_CAPTURE_STATUS
I do not know why is sent the MAV_CMD_REQUEST_CAMERA_CAPTURE_STATUS before the user trigger a picture. 
Do you have any hints? Someone else having this problem?

DonLakeFlyer, 2020-02-27:
@dogmaphobic Bumping this to a 4.0 problem.

FlavioTonelli, 2020-02-28:
Seems QGC get sucked when it does not receive the CAMERA_CAPTURE_STATUS from the camera.
More in details:
1) QGC send a MAV_CMD_REQUEST_CAMERA_CAPTURE_STATUS
2) Camera send a CMD_ACK
3) QGC stop the retransmission logic of the MAV_CMD because it received an ack.
4) Camera send CAMERA_CAPTURE_STATUS message but the message get lost (bad radio link transmission)

closed 2026-10-04 by github-actions[bot]