mavlink/qgroundcontrol #7331: Poor performance in video decoding on qgroundcontrol android

myeabhishek, 2019-03-29:
Hello Team,
I am using stable build of qgroundcontrol on android 7 downloaded from playstore. I am using it for displaying video from UDP source. I am getting extremely poor results in terms on delay and jitters. The normal delay is between 3-5 seconds along with jitters.

My guess is that on android QGC is not using hardware acceleration for h264 decoding because of which I am getting poor results on my tablet. 

 Could you please shed some light on this issue and how this can be fixed?

Yannicflight, 2019-04-01:
The same is applicable for the Ipad version, Here also very poor video when using UDP stream.

kittikhunq, 2019-10-12:
I this issue too. There a lot of defects in picture when decoding RTSP with H.264 in QGC on Android device.
When checking with other RTSP viewer program everthing was fine.
Wondering that it didn't use hardware to decode the stream.

HTRamsey, 2024-06-02:
Should be much better in master

closed 2024-06-02 by HTRamsey