mavlink/qgroundcontrol #7338: QGC does not show Video

RohanKurane, 2019-04-02:
Hi,

This is a bit high priority for me so any help would be greatly appreciated. 

I build a gst pipeline 

snprintf( str_pipeline, sizeof( str_pipeline ), "appsrc name=mysource ! videoconvert ! " "video/x-raw,width=640,height=480,format=NV12 ! " "udpsink host=%s port=5600", "192.168.x.xxx");
pipeline = gst_parse_launch( str_pipeline, NULL );

Then run in a loop via callback function to wait for frames and push it to QGC - 
 ....
    rs_wait_for_frames( camera, NULL);

    // Pull the depth and RGB data from the LibRealSense streams
uint16_t    *depth  = ( uint16_t* )   rs_get_frame_data( camera, RS_STREAM_DEPTH, NULL );
struct rgb  *rgb    = ( struct rgb* ) rs_get_frame_data( camera, RS_STREAM_COLOR, NULL );

GstBuffer *buffer = gst_buffer_new_wrapped_full( ( GstMemoryFlags )0, rgb, SIZE, 0, SIZE, NULL, NULL );

    for( int i = 0, end = WIDTH * HEIGHT; i < end; ++i )
    {
        rgb[ i ].r = depth[ i ] * 60 /  ONE_METER ;
        rgb[ i ].g /= 4;
        rgb[ i ].b /= 4;
    }
    g_signal_emit_by_name( appsrc, "push-buffer", buffer, &ret );
}

ret returned above is 0 and I can see it going on a loop when I add printfs. 

What can I be doing wrong that the video does not appear on QGC on my laptop ? 
The QGC options are selected for UDP streaming and port 5600. 

@lbegani @lucasdemarchi @LorenzMeier

RohanKurane, 2019-04-02:
There are some logs in /Documents/QGroundControl but Im not sure what to make of them. Here is a bit from 
~/Documents/QGroundControl/Logs/gst/pipeline-playing.dot
  rtp_h264_depacketizer_0x2ee6890_src_0x2ef8840 -> h264_parser_0x887faa0_sink_0x2ef9180 [labeldistance="10", labelangle="0", label="                                                  ", taillabel="video/x-h264\l       stream-format: avc\l           alignment: au\lvideo/x-h264\l       stream-format: byte-stream\l           alignment: { (string)nal, (stri... }\l", headlabel="video/x-h264\l"]
  subgraph cluster_udp_source_0x33a08d0 {
    fontname="Bitstream Vera Sans";
    fontsize="8";
    style="filled,rounded";
    color=black;
    label="GstUDPSrc\nudp-source\n[>]\nparent=(GstPipeline) receiver\ndo-timestamp=TRUE\nport=5600\nuri=\"udp://0.0.0.0:5600\"\ncaps=application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(st…\nused-socket=((GSocket*) 0x2f13520)";
    subgraph cluster_udp_source_0x33a08d0_src {
      label="";
      style="invis";
      udp_source_0x33a08d0_src_0x2ef83a0 [color=black, fillcolor="#ffaaaa", label="src\n[>][bfb][T]", height="0.2", style="filled,solid"];
    }

    fillcolor="#ffaaaa";
  }

  udp_source_0x33a08d0_src_0x2ef83a0 -> rtp_h264_depacketizer_0x2ee6890_sink_0x2ef93d0 [label="application/x-rtp\l               media: video\l          clock-rate: 90000\l       encoding-name: H264\l"]

I do NOT have the following in my gst pipeline
vaapih264enc ! h264parse ! rtph264pay !

I get a segfault when I try to include the above trying to load the plugin for vaapi.
Althought gst-inspect-1.0 vaapi shows it is loaded
/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstvaapi.so

jaxxzer, 2019-04-05:
This is not a problem with QGC.

@RohanKurane, as you have noted, it is an issue with your pipeline and the format of your stream. In order to test your application, you should use the pipeline in the end of [this section](https://github.com/mavlink/qgroundcontrol/tree/master/src/VideoStreaming#pipeline) of the documentation:
`gst-launch-1.0 udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' ! rtph264depay ! avdec_h264 ! autovideosink fps-update-interval=1000 sync=false`

When your application works with that pipeline, it will work with QGC.

closed 2019-07-09 by RohanKurane