mavlink/qgroundcontrol #7339: "No Internet Access"

RedTwenty, 2019-04-03:
## Expected Behavior
QGC should be able to connect to the internet to download maps in particular.

## Current Behavior
Laptop is connected to home wifi as normal. Full internet connection and able to surf web, receive email etc.
Open QGC and maps not loading. Check console and get "No Internet Access" message repeating. Check internet browser and no internet connection present.
Close QGC and internet connection is restored.

## Steps to Reproduce:
Can only reproduce on this laptop. Desktop computer does not have the same issue (HP Pavilion, also running Windows 10 and connected to same home wifi).

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: V3.5
- QGC build: Stable
- Flight Controller: N/A
- Autopilot (with version): N/A

## Detailed Description
Issue is present whether autopilot is connected or not.
Issue also occurs when connected to other wifi source (i.e. wifi hotspot on phone when in field).
Found old thread #5435 that appears to detail this issue in V3.2. I have rolled back to V3.1.3 and this issue is no longer present.

## Log Files and Screenshots
Console log attached.
[3-4-19_NO-INTERNET-ACCESS.txt](https://github.com/mavlink/qgroundcontrol/files/3036358/3-4-19_NO-INTERNET-ACCESS.txt)

kikislater, 2019-04-03:
Link to discussion @RedTwenty open on discuss : 
http://discuss.px4.io/t/qgroundcontrol-kills-internet/9998/

DonLakeFlyer, 2019-04-16:
@dogmaphobic Can you take a look?

DonLakeFlyer, 2020-02-08:
On the original thread the user say it went away. That said if it was a generic problem I think we'd be seeing people complaining all over the place. Likelt hardware or OS setup specific to some obscure setup.

closed 2020-02-08 by DonLakeFlyer