mavlink/qgroundcontrol #7340: .plan files should be opened via intent

ahmadbashir1612, 2019-04-03:
Intent mIntent = new Intent(Intent.ACTION_VIEW);

File file=new File(Environment.getExternalStorageDirectory()+"/QGroundControl/Missions","Bijoy.plan");

if(file.exists()){
        Uri uri = Uri.fromFile(file);
        mIntent.setDataAndType(uri, "*/*");
        mIntent.setPackage("org.mavlink.qgroundcontrol");
        startActivity(Intent.createChooser(mIntent, "View mission"));
        }

DonLakeFlyer, 2019-04-16:
I don't understand what the bug is here? Is there something that goes wrong which using a regular version of QGC? Can you explain in terms of the release version of QGC with repro steps.

ahmadbashir1612, 2019-04-16:
I tested 3.5.0 version of the app. It's not actually bug.  I wanted to mean that i tried to load a specific plan file directly from another android app( developed by me ) using the mentioned code ( implicit intent ) and that's not opening. Can this feature be implemented ?

DonLakeFlyer, 2019-04-16:
That will likely require some special permission marking in your own version of QGC. Not quite sure how to do that.

closed 2019-04-16 by DonLakeFlyer