mavlink/qgroundcontrol #7349: Not connecting to telemetry sik radio

laurent1784, 2019-04-10:
Hi everyone. I am having issues with my quadcopter. It is running on pixhawk 2.4.8 and I am using a 915 MHz stock Bluetooth telemetry set. I try yo connect to the remote radio using bluetooth and I get a heartbeat on both radios but qgroundcontrol keeps waiting for vehicle and nothing happens. Tried doing via USB and also had issues with the serial port until it finally popped. Problem is I am not receiving any information from the pxhawk? Shall I have the radio on and the drone armed or it should be getting the data from the FC anyways? I tried switching to telemetry 2 port and still having the same issue. Any help is greatly appreciated. I tried with my tablet and smartphone and seems there is no way to communicate with the pixhawk via telemetry.

laurent1784, 2019-04-10:
I tried using the iSIk Radio configurator and noticed an error appearing on the log.  Is it something I have wrongly configured?  I checked the baud rates, and parameters on Mission planner, reversed the Rx and Tx wires on the drone radio (connected to pixhawk) but still cannot get it to work.  Any help is greatly appreciated.
[image]
[image]
[image]
[image]
[image]

DonLakeFlyer, 2019-04-19:
You might want to try on one of the PX4 forums or maybe PX4 Slack channels. Since you only have QGC people here and this isn't really a QGC problem.

Hard to say what is going wrong. Can these radios connect directly via USB as well as bluetooth. If they can do USB. Then first thing would be to try USB. Make sure QGC is connecting to the radio. You can use LinkManagerLog logging to see if that is working: https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html

closed 2026-10-04 by github-actions[bot]