mavlink/qgroundcontrol #7350: UI missing not showing RC indication when RSSI not available

MaEtUgR, 2019-04-12:
## Expected Behavior
When RC is available the indicator in the top bar should light up.

## Current Behavior
If there is RC input available through an Frsky or Crossfire receiver you can pilot the drone but the RC indicator stays greyed out.

## Steps to Reproduce:
1. Have QGC opened on the PC or tablet
2. Connect the drone to the groundstation device over USB or pixracer Wifi module
3. Turn on the RC which is bound to the vehicle
4. RC works but the RC indicator stays grey

## Detailed Description
I'm assuming according to my tests that as soon as there's an RC_CHANNELS.rssi value available over MAVLink which is not 255, QGC indicates there's an RC and shows the RSSI value as percentage (should probably be in the range 0-100). Now if you're using Frsky or Crossfire with PX4 it seems the RSSI is not supported at least the sent out value is 255 for `invalid/unknown` and the indicator doesn't show if you have an RC available.

I put this issue here because I want to first here the input of e.g. @dogmaphobic or other people that worked on this topic before to know if there is already a solution or what they would expect as input from the flight control/MAVLink specification. I think the indicator for RC should also work if there's no RSSI available and it should be clear from the UI that there is not RSSI (no quality bars). And for the message definition:
- Should the autopilot not send any RC_CHANNELS message anymore if there's no RC?
- Should the autopilot still send RC_CHANNELS but leave the time_boot_ms field on the timestamp the last RC signal was received?
- Should the RSSI only be 255 if there's no RC and 254 if the RSSI is unkown/not supported but there is RC.
- Is there any other flag in a different messge that shows if RC is available or RC is lost?

## Log Files and Screenshots
[image]

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: daily 2019-04-05 2f5324fd1
- Flight Controller: Pixhawk 4
- Autopilot (with version): PX4 master 3135f9f0d2c

dogmaphobic, 2019-04-12:
I have not used a regular RC in quite some time. I need to set up a testing environment to see what's going on.

In the mean time, walking backwards, the indicator is active if the rc RSSI value is > 0 and <= 100:
https://github.com/mavlink/qgroundcontrol/blob/master/src/ui/toolbar/RCRSSIIndicator.qml#L30

Here is where it is processed:
https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L2704-L2728

Here is where it is broadcasted:
https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L1808
https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L1853

Which respectively come from `MAVLINK_MSG_ID_RC_CHANNELS` and `MAVLINK_MSG_ID_RC_CHANNELS_RAW`:
https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L689-L694

In other words, the indicator will only be active if the RSSI value is > 0 and <= 100. Your screen grab above shows a value of 255, indicating it does not have RSSI at all.

> • Should the autopilot not send any RC_CHANNELS message anymore if there's no RC?
> • Should the autopilot still send RC_CHANNELS but leave the time_boot_ms field on the timestamp the last RC signal was received?

I can't right off the top of my head answer that. Logic would indicate if you have no RC, I would not expect to receive RC_CHANNELS messages but that would need to be investigated to see if there are any ramifications.

> Should the RSSI only be 255 if there's no RC and 254 if the RSSI is unkown/not supported but there is RC.

Currently, anything other than 1-100 is considered absence.

> Is there any other flag in a different messge that shows if RC is available or RC is lost?

Not that I am aware of.

MaEtUgR, 2019-04-12:
Thanks for clarifying the specs. So I'll try to find a MAVLink spec solution. If you have any good ideas I'm open to them.

Probably the easiest would be to just send out the RC available flag the autopilot anyways has but probably not in the `RC_CHANNELS` message because that doesn't need to be sent without any RC data being available.

petertorelli, 2021-07-02:
Hi @dogmaphobic,

Your links to MAVLINK_MSG_ID_RC_CHANNELS(_RAW) are invalid, and the latter has been removed. Ignoring that, If I trace back MACLINK_MSG_ID_RC_CHANNELS to `Vehicle::_handleRCChannels`, there is a call to `mavlink_msg_rc_channels_decode`, which turns up no implementation when I grep the main branch. Weird?

```Bash
% find . -type f -print0 | xargs -0 grep mavlink_msg_rc_channels_decode
./src/Vehicle/Vehicle.cc:    mavlink_msg_rc_channels_decode(&message, &channels);
./src/FirmwarePlugin/APM/APMFirmwarePlugin.cc:        mavlink_msg_rc_channels_decode(message, &channels);
```

So my toplevel question is: if an RC transmitter is sending RSSI on a channel to the RX (which I am doing*), how does QGroundControl figure out which channel RSSI is on if there is no configuration parameter? I'm sending it back on channel 8, and I can see it in the "Radio" panel as the radio moves near/far away. How do I connect this to RC RSSI in the display window?

* I'm using a Pixhawk4 Mini with the latest QGC on the latest macOS, and an R-XSR (SBus mode connected to PX4 RCIN) sending RSSI to a Tanaris X7 Q, which then sends it back to the RC on channel 8. Seems a bit odd that I can just send it from the R-XSR to the RCIN pin, but that's a different issue.

DonLakeFlyer, 2021-07-02:
RSSI comes from the mavlink message value RC_CHANNELS.rssi which is sent by the Vehicle. It's up to the firmware as to how it fills in that rssi value in the message.

petertorelli, 2021-07-02:
Ah, thanks. I accidentally mixied up PX4-Autopilot and qgroundcontrol repositories.

closed 2026-10-04 by github-actions[bot]