mavlink/qgroundcontrol #7354: Feature request: Simplify "Action" button in Guided toolbar (Fly toolbar)

Antiheavy, 2019-04-16:
The "Action" button usually only hides 1 action underneath (at most has 2 actions one of which is "Change Altitude").   This means the most likely "action" in any context is hidden from the user, requires unnecessary extra clicks, and isn't easily "discoverable" by new users.  Most of our new users find it very confusing to find the actions under the Action button, especially when they accidentally dismiss the confirmation slider.

Request 1:
 Change the Fly toolbar to have a contextual button available depending upon the situation. The "Action" button already hides contextual sliders under it, how about we just replace the Action button itself with contextual buttons that display the appropriate slider?

Request 2:
An important related request is to eliminate the "Change Altitude" action button and replace it with an always shown right-hand altitude slider when in Hold/Pause mode.  This ensures there is only ever 1 "contextual" button. 

See the following concept graphics:
[image]

@DonLakeFlyer suggested I post this here for discussion with @dogmaphobic and others working on QGC UI.

Antiheavy, 2019-04-16:
Related to "Request 2":
There are other reasons I think Change Altitude slider should always be shown when in Hold/Pause mode. For example, you have the ability to move in X-Y by clicking around on the map and confirming, you should also have the ability to change altitude at this UI level without needing to discover it under a buried menu.

Somewhat related discussions:
mavlink/qgroundcontrol#5826
mavlink/qgroundcontrol#5748
mavlink/qgroundcontrol#6721
mavlink/qgroundcontrol#5795

Antiheavy, 2019-07-22:
@dogmaphobic or @DonLakeFlyer , any other thoughts on this proposal to simplify the "Action" button behavior?

DonLakeFlyer, 2019-07-25:
Ok so the possible actions which can show up under "Action" are:
* Start Mission
* Resume Mission
* Change Alt
* Land Abort

Reason for the Action button:
* Handle the case where multiple of these needed to be displayed at the same time
* Get the Actions back which you may have closed by using the X on the prompt slider

The only case for multiple actions appropriate at the same time are in combination with Change Altitude. It's think it's doable to show this automatically if the vehicle is "Paused". But it would require the confirmation to move to be within that alt ui as opposed to using the Action slider. Since you shouldn't be able to dismiss it without a way to get it back. All doable.

With that change then I believe you are only left with mutually exclusive commands which could be displayed as the last button. Which would only then be needed if you closed the contextual slider when it popped up and wanted it back.

Unlikely to get to this before I head out of town again. But I'll start on it when I get back. Quite complex change so will need to happen on master.

Antiheavy, 2019-07-25:
> Since you shouldn't be able to dismiss it without a way to get it back.

If you drag the change altitude column up or down, the confirmation slider should automatically appear at the bottom of the screen. If you dismiss the confirmation slider then whatever change you made on the change altitude column should revert.  To bring back the confirmation slider you'd have to make another change on the change altitude column with which needs confirming.

> Unlikely to get to this before I head out

no problem.  Users aren't screaming for it, but in general hiding things under the "Action" button is a point of confusion and mystery to them.  Putting more functionality right at the top level UI makes things more discoverable and intuitive.

DonLakeFlyer, 2019-07-26:
Awesome way to handle the change alt confirm. I'll do that.

dakejahl, 2021-02-26:
Any updates here? If I started working to implement these changes, would I be stepping on anyone's toes?

DonLakeFlyer, 2021-03-02:
> Any updates here?

Nope. Still seems like a reasonable thing to do.

Antiheavy, 2021-03-02:
> If I started working to implement these changes, would I be stepping on anyone's toes?

@dakejahl let's give it a try!