mavlink/qgroundcontrol #7358: Add Follow and Simple modes to list of available modes for ArduPilot Rover

rmackay9, 2019-04-17:
ArduPilot Rover-3.5 (the current stable release) has [Follow](http://ardupilot.org/rover/docs/follow-mode.html) and Simple modes but these don't appear on the list of available flight modes in QGC.

Could we get these two added?  Follow is "6" and Simple is "7" ([see mode.h here](https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/mode.h#L28))

Thanks!

DonLakeFlyer, 2019-04-17:
@rmackay9 Simple should have already been there. But Follow is not because ArduPilot follow is not implemented. Someone else added support for this on PX4 side a while back. I should be able to make it work for ArduPilot as well. Does ArduPilot follow work based on the FOLLOW_TARGET mavlink message?

rmackay9, 2019-04-17:
@DonLakeFlyer, thanks for the response.  it's on our to-do list to add support for the FOLLOW_TARGET message but at the moment Follow is just another flight mode.  It will automatically grab onto the first non-GCS sysid it sees and follows at the current distance so in the best case it doesn't require any parameter changes from the user nor any extra info from the GCS.

DonLakeFlyer, 2019-04-18:
@rmackay9 Just so I understand correctly. It looks at telemetry from another vehicle then to follow from? QGC isn't really involved?

I'll add it Follow.

rmackay9, 2019-04-18:
@DonLakeFlyer, yes, that's right.  thanks!

DonLakeFlyer, 2019-04-19:
OK, all done then

closed 2019-04-19 by DonLakeFlyer