mavlink/qgroundcontrol #7360: QGC not able to flash NXP FMUK66

LeoMustafaNXP, 2019-04-17:
## Expected Behavior
QGC should be able to connect and flash new firmware to NXP FMU K66.

## Current Behavior
Connection is only possible after driver update: 
How to update the driver is described in: https://nxp.gitbook.io/hovergames/userguide/qgroundcontrol/windows-drivers

On Windows 7/10 QGroundcontrol chancel the flashing progress with a time out.
On Mac OS the flashing progress doesn't start. 

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. update the driver as descirbed in : https://nxp.gitbook.io/hovergames/userguide/qgroundcontrol/windows-drivers 
2. start QGC
3. open the flashing tab
4. connect the NXP FMU K66 over USB

## System Information
When posting bug reports, include the following information
- Operating System: Windows 7/10; macOS Mojave 10.14.3
- QGC Version: 3.5.0
- QGC build: stable
- Flight Controller: NXP FMU K66
- Autopilot (with version): PX4 1.8.1

## Detailed Description
At first the QGC driver must be updated with the right product and vendor ID: 
take a look at the following descibtion: https://nxp.gitbook.io/hovergames/userguide/qgroundcontrol/windows-drivers

At second QGC should be able to flash the firmware on the NXP FMU K66.

## Log Files and Screenshots
[image]

davids5, 2019-04-17:
@LeoMustafaNXP Please run [usbview-v2](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview) and capture the bootloader and app's usb info and post them here.
To get the bootloader use "reboot -b" on the console.

jarivanewijk, 2019-04-19:
I'm not @LeoMustafaNXP, but I also have this issue so I'll jump in as well. This is what I was able to capture with usbview. I hope this is what you need, @davids5. :)

[app.txt](https://github.com/mavlink/qgroundcontrol/files/3097527/app.txt)
[bl.txt](https://github.com/mavlink/qgroundcontrol/files/3097528/bl.txt)

System info:
- Windows 7
- QGroundControl daily build, commit c0ea93e95
- Hardware: NXP FMUK66
- PX4 Firmware master branch, commit 2b68076
- PX4 Bootloader master branch, commit 5b973aa

Also still had to modify the PX4 drivers on Windows to connect to the FMU, as mentioned here: https://nxp.gitbook.io/hovergames/userguide/qgroundcontrol/windows-drivers
Should we report this again / open another issue about this?

Also, on the same system when I start a Ubuntu virtual machine, pass through USB to the VM, and start QGC inside it, firmware upload works fine from the virtual machine.

igalloway, 2019-04-19:
@davids5 One part of this issue is that we did update the PX4 Drivers with the new VID and PID, but don't know how to get them included in the QGC install. BUT in addition to that - on windows only - we still seem to be "blowing by" the bootloader recognition stage in QGC. From what i recall (it was a little while ago when i tested) FMU would be in bootloader mode (flashing yellow) and you could get into the bootloader - custom firmware - if you clicked really fast, but then QGC seemed like it didn't see the device in bootloader mode and quickly timed out before any bootloading happened. So it kinda gets to bootloader in QGC, but then fails to stay in that mode.

DonLakeFlyer, 2019-04-20:
QGC needs to know the VID/PID (compiled in) or it may fallback to some string matching in order to flash. Placing the board in bootloader mode won't help if QGC doesn't recognize the board.

LeoMustafaNXP, 2019-04-23:
@davids5 attached you find the files captured with usbview. 
[USBView_FMUK66_Bootloader_Leo.TXT](https://github.com/mavlink/qgroundcontrol/files/3107320/USBView_FMUK66_Bootloader_Leo.TXT)
[USBView_FMUK66_Leo.TXT](https://github.com/mavlink/qgroundcontrol/files/3107321/USBView_FMUK66_Leo.TXT)

davids5, 2019-04-23:
Thank you @LeoMustafaNXP & @jarivanewijk 

@DonLakeFlyer The VID is NXP's 0x1FC9 and the PID is 0x1c. Where is the VID checked in the code and is it just 3DR's VID that is checked for?

jarivanewijk, 2019-04-24:
@davids5 @DonLakeFlyer 

I found these two VID/PID definitions:
[/src/comm/USBBoardInfo.json](../blob/master/src/comm/USBBoardInfo.json)
[/src/VehicleSetup/Bootloader.h](../blob/master/src/VehicleSetup/Bootloader.h)

In both VID and PID seem to be correct for NXPhlite. Also, if they were not defined correctly, wouldn't that also cause issues on all operating systems? We have no issues on Linux.

davids5, 2019-04-24:
@jarivanewijk  As @igalloway says:  Is windows 10 Just not enumerating fast enough? A quick test would be to reboot and stay in the bootloader. On the console `reboot -b" Second thing to now is the description has changed.

Bootloader is "PX4 BL NXPHlite v3.x"
App is            "PX4 FMUK66 v3.x"

DonLakeFlyer, 2019-04-24:
> Also, if they were not defined correctly, wouldn't that also cause issues on all operating systems?

That is correct. A missing VID/PID will cause QGC to not recognize the board. It will not cause the board to fail at bootloader sync.

jarivanewijk, 2019-04-26:
@davids5 I did use "reboot -b" command when logging the USB info with usbview. At the time, I didn't think about checking if firmware upgrade would work when I rebooted the board into bootloader this way. I think this is what you suggested to check as well?

So, I gave it a try now. Still doesn't work. With or without external power. See error below.
[image]

I'm not sure if there is anything else we can log or check?

davids5, 2019-04-30:
@igalloway @LeoMustafaNXP @jarivanewijk - I started looking  at this today. 2 issues I saw are the a) the Name and b) more important the serial number. The change in serial number causes the the device to enum on COMx for the bootloader and COMy for the application.  After fixing that I see consistent enumerations but today's daily of QGC dose not work on my Windows set up with ANY flight controller. I will reinstall and retest in the AM and let you know what I find

jarivanewijk, 2019-05-03:
Bootloader side seems to be fixed now, thanks @davids5! Firmware upgrade (custom firmware option) through QGroundControl is working on both Windows and Linux. @LeoMustafaNXP - can you also try on macOS?

What is still not working is using the Stable / Beta / Developer builds downloaded by QGroundControl. It will give the error "Unable to find specified firmware download location". Seems like the PX4 builds for FMUK66 are not automatically being uploaded to the right location? Custom firmware (supplying the file yourself) works fine, by the way.

Also, the modified Windows driver still needs to be integrated. @igalloway - I think you know more about this? You asked somebody to modify the .inf files and create a new certificate, if I remember it correctly?

LeoMustafaNXP, 2019-05-03:
The new Bootloader seems not to work on macOS. I can not flash the FMU over QGC. The Bootloader is not recognized in QGC.

davids5, 2019-05-03:
Thank you @jarivanewijk and @LeoMustafaNXP !

I will look at it on the Mac next week.

> What is still not working is using the Stable / Beta / Developer builds downloaded by QGroundControl. It will give the error "Unable to find specified firmware download location". Seems like the PX4 [...]

see https://github.com/PX4/Firmware/pull/11953

igalloway, 2019-05-03:
> see PX4/Firmware#11953 <https://github.com/PX4/Firmware/pull/11953>

Hi all
We had the .inf files modified and signed. How do we get this integrated
into the QGC distribution? Leo could you send/post the files?

LeoMustafaNXP, 2019-05-03:
Attached are the modified and signed .inf files.
[cert_procedure.zip](https://github.com/mavlink/qgroundcontrol/files/3141849/cert_procedure.zip)

jarivanewijk, 2019-05-07:
@DonLakeFlyer - Could you maybe help us? 

The NXP FMUK66 VID/PID are not yet integrated into the PX4 Windows drivers that are included with the QGroundControl installer. Months ago, NXP was asked to provide instructions or sign the modified driver. I don't know the details of what was asked and why, but I think @LeoMustafaNXP provided the required files.
> Attached are the modified and signed .inf files.
> [cert_procedure.zip](https://github.com/mavlink/qgroundcontrol/files/3141849/cert_procedure.zip)

I think we are lost. What do we need to do to get this integrated?

DonLakeFlyer, 2019-05-07:
QGC uses the Windows drivers which are built and come from Mission Planner. You'll need to talk with the Mission Planner folks about getting them updated. And then let me know when that is done and I'll pull a new version from MP.

Junkim3DR, 2019-05-10:
### Tested on NXP FMUK66 with LJI X4 500 frame

**Procedure:** 
- Windows
1 - Updated drivers as described in: https://nxp.gitbook.io/hovergames/userguide/qgroundcontrol/windows-drivers
2 - Opened QGC and attempted to flash FUK66 with Master firmware, but "Upgrade Cancelled" message popped up.
3 - Repeated procedure 3 times, but same result every time.
-Mac
1 - Tried to flash FUK66 with Master firmware, "Upgrade Cancelled" message popped up.
2 - Uninstalled and reinstalled QGC in order to flash Master again and got same results.

**Relevant Information**
Operating System: Windows 10; macOS Mojave 10.14.4
QGC Version: 3.5.0
QGC build: Master Commit (e80d3940)
Flight Controller: NXP FMU K66

### Test Flight on NXP FMUK66 with LJI X4 500 frame

**Note:** Downloaded Master from Jenkins and loaded to NXP FMUK66 as Custom Firmware.

**Modes Tested:**
- Position Mode: Good.
- Altitude Mode: Good.
- Stabilized Mode: Good.

- Procedure 
Arm and Take off in position mode, after flying for approximately one minute, switched to altitude then stabilized mode.

**Log:**
https://review.px4.io/plot_app?log=95cf27d8-4e91-49a4-b9a7-8b9ee49465c6

**Video:**
https://youtu.be/g5sx7aXyj0o

davids5, 2019-05-12:
@Junkim3DR you will need to replace the bootloader with https://github.com/PX4/Bootloader/pull/133 @igalloway  FYI.

davids5, 2019-05-12:
> I will look at it on the Mac next week.

@LeoMustafaNXP I tested on Mac mini. 
[image]

[image]

Please insure you updated the bootloader and retest and post images of results here.

LeoMustafaNXP, 2019-05-13:
@davids5 now it works also on MacOS. Great work.

Junkim3DR, 2019-05-20:
### Tested on NXP FMUK66 with LJI X4 500 frame

**Procedure:** 
- Tested on Mac (MacOS Mojave v10.14.4).
1 - Uninstalled QGC.
2- Rebooted Macbook.
3- Installed the latest QGC version from http://qgroundcontrol.com.
4- Rebooted Macbook.
5- Opened QGC and selected Firmware tab under Vehicle Setup.
6- Connected NXP board to Macbook with USB cable.
7- Once Firmware Setup popped up, selected Advanced Settings the proceeded to choose Developer Build (Master).
8- Clicked OK to trigger flashing process.

**Notes:**
Able to upload Master to NXP board from QGC.

**Relevant Information**
Operating System: macOS Mojave 10.14.4
QGC Version: 49ef8c141
Flight Controller: NXP FMU K66

jarivanewijk, 2019-06-12:
@DonLakeFlyer The Mission Planner drivers for Windows have been updated. https://github.com/ArduPilot/MissionPlanner/pull/2155
I think once the driver is also included in QGroundControl, this issue is completely solved.

FYI @igalloway @LeoMustafaNXP @davids5

jarivanewijk, 2019-07-10:
@DonLakeFlyer any progress on getting the new drivers into QGC?

It would be great if we could get a stable QGC release with FMUK66 support. That will be very useful for the participants of the HoverGames. From what I have read, it seems unlikely that there will be a new major stable release soon, but maybe it would be possible to do a 3.5.4 release with the updated drivers and these two pull requests: 

https://github.com/mavlink/qgroundcontrol/pull/7409 (rename NXPhlite to FMUK66)
https://github.com/mavlink/qgroundcontrol/pull/7423 (define PX4 firmware URLs for FMUK66)

DonLakeFlyer, 2019-07-16:
Sorry I've been out of town. I'll look at getting to this when I dig myself out form under backlog.

DonLakeFlyer, 2019-07-18:
@jarivanewijk Can you check the pull to make sure I goet everything you wanted from master?

jarivanewijk, 2019-07-21:
@DonLakeFlyer I think that is all, yes. Only the updated USB driver still needs to be pulled in from Mission Planner, right?

closed 2019-07-24 by DonLakeFlyer

davids5, 2019-07-24:
Thank you @DonLakeFlyer !

jarivanewijk, 2019-07-27:
Thanks a lot, @DonLakeFlyer !

DonLakeFlyer, 2019-07-27:
I marked new Stables yesterday. So this should be out there now.

igalloway, 2019-07-27:
Thanks three times!