mavlink/qgroundcontrol #7367: QGC v3.5.0 Won't Open in Windows 10 Pro

rollysing, 2019-04-19:
Installed QGC v3.5.0 in Lenovo laptop running Windows 10 Pro v1803. QGC attempts to open but fails. While looking at Task Manager, the app starts to open but Status momentarily shows Suspended, then QGC disappears.
- Have disabled signature requirement during startup prior to installation.
- Have tried running Safe Mode and Compatibility versions but still no joy.
- Tried installing the latest Daily Build and am got the same result as described above.
- With some registry modifications, installed it in Windows Safe Mode. The app opened but because there's no network connection, it didn't download any maps. After rebooting to normal startup, QGC crashes as well.

Here is a short video of what happens in the Process tab of Task Manager:
https://youtu.be/HfB_WZuV08Q

The crash dumps can be downloaded here: https://1drv.ms/f/s!Ar66Lk6yrxscgjKW3SjU_oPLtz4p

Any ideas/ advise on how to resolve is highly appreciated.

DonLakeFlyer, 2019-04-21:
Looks like a graphics driver crash. Have you tried to Windows QGC Graphics Compatability modes:

https://docs.qgroundcontrol.com/en/Support/troubleshooting_qgc.html

rollysing, 2019-04-22:
Yes, all of them fail the same way. I've also tried the different options in Windows Compatibility and none of them worked.

rollysing, 2019-04-27:
@DonLakeFlyer 

v3.5.1 Crash Dumps
https://1drv.ms/f/s!Ar66Lk6yrxscgjKW3SjU_oPLtz4p

rollysing, 2019-05-04:
@DonLakeFlyer Thinking about some more of graphics driver crash. Here are the specs of the laptop/tablet display. FYI... Mission Planner installs and runs without any issue.

- Intel HD Graphics 515 Display adapter with Driver Date 2/7/2019 Driver Version 25.20.100.6576
- Resolution: 2160 x 1440
- Text size: 175%

Hope this helps to point to possible resolution.

Thanks.

DonLakeFlyer, 2019-05-04:
Yeah Intel graphics cards are usually the ones which cause problems. Their OpenGL driver sucks. Which is why those additional statup modes were added. Which have always worked in the past. Can you capture crash dumps from those two additional startup modes as well. Maybe those broke somehow.

rollysing, 2019-05-05:
@DonLakeFlyer Here are the Compatibility and Safe Mode crash dumps. Hope this points you to the right direction for the resolution.
https://1drv.ms/f/s!Ar66Lk6yrxscgjKW3SjU_oPLtz4p

DonLakeFlyer, 2019-05-07:
@dogmaphobic Do you have new qgc4 windows builds available? I'd be curious to see if full native qml works around these OpenGL windows crashes.

dogmaphobic, 2019-05-07:
My Windows build system got hosed the last time I "upgraded" Qt. I need to get that fixed. I'll let you know...

DonLakeFlyer, 2019-05-07:
If you could get an installer to @rollysing to try that would be great.

dogmaphobic, 2019-05-07:
@rollysing, here we go: https://drive.google.com/file/d/1cqnzPrZfFnyiWgybXwaxQ7UQJ6ImCJtb/view?usp=sharing

Note that this is a new, under development branch. It matches master but certain things are missing (such as log replay, charting, etc.)

rollysing, 2019-05-07:
@dogmaphobic @DonLakeFlyer Nope, it was the same as the regular install. Here are the crash dumps of all 3 program versions.
https://1drv.ms/f/s!Ar66Lk6yrxscgj9bneTfx42JAJiC

rollysing, 2019-06-13:
Just following up to see if there's been any progress on this.

Thanks.

DonLakeFlyer, 2019-06-18:
@rollysing Can you try this new version: https://qgroundcontrol.s3-us-west-2.amazonaws.com/DonLakeFlyer/QGroundControl-installer.exe

rollysing, 2019-06-18:
No joy. Tried all three startups.

Thanks.

DonLakeFlyer, 2019-06-19:
Not sure we are going to be able to fix this until we move to 64 bit and Qt 12. Then we may need to get a fix out of Qt for it. Possibly has something to do with suspending. Not sure if there if a way to make that not happen by change the power settings on the laptop or not.

rollysing, 2019-06-19:
Ok, thanks. So is that the difference of QGC vs Mission Planner? Just curious since MP installs, runs and upgrades without any issue.

Thanks for all the attention and assistance.

DonLakeFlyer, 2019-06-19:
MP and QGC are radically different. MP is a native windows C# application. QGC is a cross platform C++ using a library (Qt) to provide the cross platform and ui support.

DonLakeFlyer, 2019-06-19:
Also in general it's the Lenovo Thinkpads which cause the most problems in combination with the Intel integrated graphics driver which appears to be crap. If you do internet searches associated with the Intel drivers you'll find a wasteland of apps having problems with them.

rollysing, 2019-06-22:
Just odd that I've been using Lenovo laptops for a better part of this decade and this is the first (and only) one that has run into this problem. Admittedly that this is the only one that has a 2160 x 1440 native resolution.

Thank you for all the help.

jaxxzer, 2019-07-03:
@rollysing what is your laptop model?

rollysing, 2019-07-03:
Lenovo Thinkpad X1 20GG001KUS

DonLakeFlyer, 2019-07-03:
@antiheavy Is this similar to your folks laptops?

DonLakeFlyer, 2019-07-03:
I want to try a few more things...

DonLakeFlyer, 2019-07-03:
@rollysing Can you try this: https://qgroundcontrol.s3-us-west-2.amazonaws.com/DonLakeFlyer/QGroundControl-installer.exe. It has gstreamer compiled out of it. If it still crashes attach the dump.

rollysing, 2019-07-03:
@DonLakeFlyer No joy. CrashLogs here https://1drv.ms/f/s!Ar66Lk6yrxscgkj63lwFDT4-1xPp

rollysing, 2019-07-03:
@DonLakeFlyer By the way, that installer was being blocked by Windows. Had to go Advanced.
[image]

DonLakeFlyer, 2019-07-04:
@rollysing Are the three crash logs in there the three startup variants? Or just three crashes from regular boot? 

As far as the windows protection thing. It always does that since the installer is not signed. @hamishwillee May be worthwhile to include in the docs to steps to work around this. OSX has a similar thing where you need to right-click and select Open to be allowed to run for the first time since it's not signed.

DonLakeFlyer, 2019-07-04:
@dogmaphobic FYI...

So the stack trace is this:
[image]

From internet search dsengine.dll seems to be related to Qt QCameraInfo: https://stackoverflow.com/questions/56180327/which-dlls-are-needed-for-qt-multimedia-qcamerainfoavailablecameras-to-retu

The version @rollysing is testing has QStreamer compiled out by hacking the build files. Wondering if even though gstreamer is complied out whether QGC still cranks up the camera stuff where it tries to get list of cameras from OS? Which may be causing the crash.

dogmaphobic, 2019-07-04:
> Wondering if even though gstreamer is complied out whether QGC still cranks up the camera stuff

Yes, UVC video is separate from GStreamer. UVC support is handled through Qt Multimedia, using a very different code path.

DonLakeFlyer, 2019-07-04:
Ok, let me hack that out and see what happens.

DonLakeFlyer, 2019-07-04:
@rollysing Can you try this one: https://qgroundcontrol.s3-us-west-2.amazonaws.com/DonLakeFlyer/QGroundControl-installer-nogstuvc.exe? This has both gst and uvc compiled out.

Antiheavy, 2019-07-04:
> @Antiheavy Is this similar to your folks laptops?

The Lenovo's that our internal folks are using and also many of our customers are using are various models in the Thinkpad P-series.  The application crashes we are experiencing for the most part sound different than the one @rollysing is reporting (although maybe all related somehow).  However, we recently had a customer who seemed to be having an issue similar to @rollysing - no idea what type of laptop they had.  They were reporting the application launching and immediately crashing, 100% repeatable.  They updated to a newer version of QGC with our custom plugin (Sentera v1.4.0 FWIW) and their app crashing problem went away.

rollysing, 2019-07-04:
> @rollysing Can you try this one: https://qgroundcontrol.s3-us-west-2.amazonaws.com/DonLakeFlyer/QGroundControl-installer-nogstuvc.exe? This has both gst and uvc compiled out.

Tally Ho! This one did it!

DonLakeFlyer, 2019-07-05:
Ok, let me start putting things back one at a time and you can try them and then hopfully we can nail it down exactly.

DonLakeFlyer, 2019-07-05:
@rollysing Let me know what happens with this one: https://qgroundcontrol.s3-us-west-2.amazonaws.com/DonLakeFlyer/QGroundControl-installer-nouvc.exe

rollysing, 2019-07-05:
@DonLakeFlyer That worked albeit, UVC device not showing in Video Source as the installer name indicated.

DonLakeFlyer, 2019-07-05:
Are you actually trying to use a UVC device on this laptop with QGC? If so, didn't realize that.

rollysing, 2019-07-05:
Yes, for OSD via FPV camera.

rollysing, 2019-07-12:
@DonLakeFlyer Just following to see if there's an updated installer that would include UVC integration.

Thanks.

rollysing, 2019-08-15:
Hello @DonLakeFlyer . Just following up to find out if there's been any progress on a version that will allow to use UVC input? Thanks.

DonLakeFlyer, 2019-12-24:
The new gsstreamer work should fix the startup problems we hope.

rollysing, 2019-12-24:
Would this be on the stable release or nightly build? I think I tried v3.5.6 and it still wouldn't open.

DonLakeFlyer, 2019-12-25:
This is a WIP that should be coming soon in a daily: https://github.com/mavlink/qgroundcontrol/pull/8093

DonLakeFlyer, 2020-03-05:
@rollysing You're working now right?

closed 2020-03-05 by DonLakeFlyer