mavlink/qgroundcontrol #7382: QGroundControl 3.5.1 won't start on Ubuntu 16.04

sfalexrog, 2019-04-27:
QGroundControl v3.5.1 for Linux (packaged as an AppImage) won't start on Ubuntu 16.04.6 LTS due to missing `libicui18n.so.56`. Current ICU package version for Ubuntu 16.04.6 is 55.

## Expected Behavior
QGroundControl should launch.

## Current Behavior
QGroundControl fails to launch with the following message:

```bash
$ ./QGroundControl.AppImage 
/tmp/.mount_4m096M/QGroundControl: error while loading shared libraries: libicui18n.so.56: cannot open shared object file: No such file or directory
```

## Steps to Reproduce:

1. Update packages on Ubuntu 16.04 to their latest versions
2. Download the QGroundControl v3.5.1 AppImage for Linux and set run permissions (`chmod +x ./QGroundControl.AppImage`)
3. Open terminal and navigate to QGroundControl.AppImage
4. Try to run the image (`./QGroundControl.AppImage`)

## System Information

- Operating System: Ubuntu 16.04.6 LTS
- QGC Version: 3.5.1
- QGC build: stable

jnunezortuno, 2019-04-27:
Also here with:

Operating System: Ubuntu 18.10 (cosmic)
QGC Version: 3.5.1

Have no problems with previous 3.5.0 version

sfalexrog, 2019-04-27:
Looks like the bundled Qt libraries link explicitly to `libicui18n.so.56`, `libicuuc.so.56` and `libicudata.so.56`. The AppImage used to contain these libraries with their version suffixes, but now it contains these libraries without the suffixes.

I've tried unpacking the AppImage and creating symlinks for the missing files (`libicudata.so.56 -> libicudata.so`, `libicui18n.so.56 -> libicui18n.so`, `libicuuc.so.56 -> libicuuc.so`), and this allowed QGroundControl to launch.

I believe this bug might have something to do with this commit: https://github.com/mavlink/qgroundcontrol/pull/7222/commits/10bf1605b5f6d52b5859bb4f160daf044a46e921 - as far as I can tell, the intention was to use system libraries if possible, but it probably should have contained extra logic for launching QGroundControl from AppImage. Maybe we could create symlinks for the missing libraries during AppImage creation?

IeiuniumLux, 2019-04-27:
I'm also using Ubuntu 16.04.6 LTS and QGC 3.5.1 just failed with the same error when I tried to launch it.

`libicui18n.so.56: cannot open shared object file: No such file or directory`

DonLakeFlyer, 2019-04-27:
So is it ok to just revert the commit listed above? The pull is here: https://github.com/mavlink/qgroundcontrol/pull/7222. If I revert that will anything else go wrong for normal usage?

Folks who want to run different libicui will need to manually symlink magic themselves I think but regular users will work fine.

bys1123, 2019-04-27:
But I tried latest version few days ago, works fine.

sfalexrog, 2019-04-27:
Just tried to build an AppImage from current master (https://github.com/mavlink/qgroundcontrol/commit/e2b3a235facd2bcfceb78f6de80f6adb19fd1d89 at the time of writing); curiously enough, it has `libicu*.so`, `libicu*.so.56` and `libicu*.so.56.1` files that are seemingly identical, and it works just fine, although it's somewhat larger than the release AppImage (possibly due to `libicu*` duplicates).

I'm at a loss; there seems to be a step that excludes some of the `libicu*` files from release, but I'm not triggering it locally for some reason.

sfalexrog, 2019-04-27:
@DonLakeFlyer, I believe this is the commit that "fixes" this bug: https://github.com/mavlink/qgroundcontrol/commit/aab353a0e032b196d3f55eb2299f7383f2c8e374 - although there will be several copies of icu libraries in the AppImage, making it somewhat larger (~100 MB vs ~75 MB).

DonLakeFlyer, 2019-04-27:
@sfalexrog Could you create a pull against Stable_V3.5.1 branch which brings that across?

DonLakeFlyer, 2019-04-27:
Also if Qt connects directly to .56 version then why include *.so in the AppImage. Qt won't reference it will it? Now that I say that I'm not conviced master is correct either.

@stefandunca @sithlord48 Including folks from original changes.

DonLakeFlyer, 2019-04-27:
I think at a minimum I'm going to revert Stable_V3.5.1 then we can figure out the right thing to do to make all people happy on master.

DonLakeFlyer, 2019-04-27:
Can one of you guys verify that this build works for you: https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/Stable_V3.5.1/QGroundControl.AppImage

maciek01, 2019-04-28:
@DonLakeFlyer   https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/Stable_V3.5.1/QGroundControl.AppImage works fine now - thanks for quick response. (18.10 LTE)

kikislater, 2019-04-28:
Same on archlinux : icu is version 64 ...

DonLakeFlyer, 2019-04-28:
@kikislater Which means what the new link I posted is not working for you? Archlinux isn't really supported as a tested build.

kikislater, 2019-04-28:
Tested on appimage 3.5.1
Shown you 3.5.2 afterwards, have to test, I will tell you ... 
A nice guy from Australia maintain packages at this time (si I run it Iwithout problem just to signal, don't be angry) but I assume appimage will be fine and does not depend in distribution

sfalexrog, 2019-04-29:
@DonLakeFlyer Sorry for taking so long. I've just tried the build you've linked, and it works on my Ubuntu 16.04 and 18.04 just fine!

I'm not quite sure what the original P/R authors wanted to accomplish. My guess is that the first P/R (the one that removed the version suffix from `libicu*`) was intended to allow developers to use their system Qt and icu builds; but it probably wouldn't produce proper builds anyway. [According to the icu packaging guide](http://userguide.icu-project.org/packaging), any application or library that wants to be linked against `libicu*.so` will actually reference `libicu*.so.VER`. I believe that the proper way to do that would be to look at the icu library name that Qt libraries reference, and copy that into the AppImage; unfortunately, I'm not familiar with qmake projects and am not sure how to do that.

DonLakeFlyer, 2019-04-30:
Thanks for reporting back. Stable works now so closing.

closed 2019-04-30 by DonLakeFlyer

Kukhokuhle, 2020-04-10:
> Can one of you guys verify that this build works for you: https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/Stable_V3.5.1/QGroundControl.AppImage

Hi, the link doesn't direct me to a QGC build. Please help?

DonLakeFlyer, 2020-04-11:
Stable 3.5.1 build is ages old, hence no longer available on S3. If you really need an old build you should be able to get it from GitHub Releases.