mavlink/qgroundcontrol #7396: QGroundControl 3.5.2 (Win10) crushes when trying to upgrade firmware

nconrad841, 2019-04-30:
## Expected Behavior
Starting QGC 3.5.2 under Win10. Pixhawk4 plugged via USB.
Going in QGC to Firmware -> unplug and afterwards plug in the Pixhawk 4 is should start to upgrade the firmware

## Current Behavior
QGC crushes after Pixhawk 4 plug in. There is no crush report for more information.

## Steps to Reproduce:
1. Plug in Pixhawk 4 via USB
2. Start QGC 3.5.2
3. Go to Firmware
4. Unplug Pixhawk 4
5. Plug in Pixhawk 4

## System Information
- Operating System: Windows 10
- QGC Version: 3.5.2
- Flight Controller:  Pixhawk 4

DonLakeFlyer, 2019-04-30:
Doesn't repro for me.

Can you get a crash dump file as shown here: https://docs.qgroundcontrol.com/en/Support/Support.html

nconrad841, 2019-05-02:
Here are some dump files of this crash.
https://drive.google.com/open?id=1lMPkhYdHl5mgtp7_BVdq0sXEDdLMC7A1