mavlink/qgroundcontrol #7397: GeoFence crash on MacOS

FlavioTonelli, 2019-04-30:
I can let QGroundControl V3.5.1 crash every time performing the following steps:
1) Open QGroundControl
2) Open the Plan View
3) Open the Fence Planning window
4) Connect a Pixhawk board running PX4 1.8.2
5) Click on the Polygon Fence button
6) Click on one of the polygon edges. The UI shall pop up a menu.
7) Close the menu (Clicking back on the map)

The application output the error QObject::connect: invalid null parameter and crash.

I tested it on both MacOS and Windows machine and I can reproduce it ONLY on the MacOS one.
[image]

DonLakeFlyer, 2019-04-30:
Crap, this looks to be a Qt bug with Menu qml.

DonLakeFlyer, 2019-04-30:
Stable build updated. Will go out when I pull the trigger on 5.3.3. Daily Build fix should be available by tomorrow.

DonLakeFlyer, 2019-04-30:
@FlavioTonelli And thank you for the bug report.

closed 2019-04-30 by DonLakeFlyer