mavlink/qgroundcontrol #7402: [iOS] Version is too old

XAS-712, 2019-05-01:
This system is for posting bugs or feature requests ONLY. For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

The iOS version in App Store is very old.
Now the release is @3.5, but iOS version in App Store is still 3.4.3.

----
When posting bug reports, include the following informaiton:
- Ground station operating system. **iOS**
- QGroundControl version and build (daily, stable, self-built from source, etc.) **From App Store **
- **If you are using a Stable build which is not the latest, first step is to install latest Stable build and try again prior to entering a bug report.** **Sorry, I cannot find any newer version. In App Store v3.4.3 is the latest version.**
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

**All this template text should be deleted before creating your issue.**

DonLakeFlyer, 2019-06-11:
@dogmaphobic Where are things at with iOS version?

HTRamsey, 2024-06-02:
Doubtful that iOS app will be updated, but you can sideload iOS apps now.

closed 2024-06-02 by HTRamsey