mavlink/qgroundcontrol #7405: No terrain data (for Finland?)

sgofferj, 2019-05-01:
## Expected Behavior
QGC should get terrain data and send it to the MAV

## Current Behavior
QGC says "Plan is waiting on terrain data from server for correct altitude values".
Console is full of "Error in fetching elevation tile. Empty response."

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Plan a mission with altitude over terrain
2. Location: N61.2556694, E24.0564945
3. Try to upload

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 18.04
- QGC Version: 3.5.1
- QGC build: stable
- Flight Controller: Pixhawk
- Autopilot (with version): Ardupilot Copter 3.6.8

## Detailed Description
I checked through the source and found that terrain data seems to be downloaded from airmap.com. My guess would be that a US company does not have elevation data for everywhere in the world. Suggestion: use public worldwide SRTM data at least as fallback and/or allow the user to manually copy .hgt files to a folder.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

DonLakeFlyer, 2019-05-01:
Can you attach a Plan file for the area without using terrain?

sgofferj, 2019-05-01:
[plan.zip](https://github.com/mavlink/qgroundcontrol/files/3135224/plan.zip)

berserk84, 2020-08-31:
I have the same problem in Norway.
But I see on page:
https://developers.airmap.com/docs/elevation-api
that airmap elevation API is not supporting "extreme" latitudes:
```
Elevation Dataset Limitations
The elevation dataset for this API is limited to latitudes from 56ºS to 60ºN
```

So it would be a nice feature to be able to use different elevation data source than airmap.com

closed 2026-10-04 by github-actions[bot]