mavlink/qgroundcontrol #7410: wrong mission time in android app

eifel2006, 2019-05-03:
[image]
[image]
As you can see, the same mission in android app 2 times longer than in pc app

eifel2006, 2019-05-03:
when you switch to mission design, 28 minutes 48 seconds are automatically calculated.

eifel2006, 2019-05-03:
I have 3.5.2 stable and last daily from 2019.05.02. The same problem - 28min 48sec in new mission

DonLakeFlyer, 2019-05-03:
Can you attach the plan file

eifel2006, 2019-05-03:
It doesn't matter. Flight plan from PC have +28.48 on android, and flight plan from android is right on PC.
Flight plan here: https://yadi.sk/d/uaAO1P3pSw0fMg

eifel2006, 2019-05-03:
I was try it on alcatel one touch idol3, xiaomi redmi 4, xiaomi redmi note 4x.