mavlink/qgroundcontrol #7412: Improve scroll-ability of log replay slider

yutaoliu, 2019-05-03:
QGroundControl version 3.5.0 (stable) running in Ubuntu 18.04.

The slider for replaying tlog has very poor time resolution i.e. I can only drag it to start the log replay from a couple of locations. @DonLakeFlyer says this is a bug and it should be able to start from any location.

DonLakeFlyer, 2019-05-03:
Can you give me a link to a telemetry log which shows this behavior? I just tried it myself and as able to set the slider all over the place.

Antiheavy, 2019-05-04:
>  I just tried it myself and as able to set the slider all over the place.

really?  The slider has never worked properly for me, always as @yutaoliu describes above.  Might it behave differently in Windows (my OS) or other versions?

Hence this related feature request: https://github.com/mavlink/qgroundcontrol/issues/5730

DonLakeFlyer, 2019-05-04:
> The slider has never worked properly for me

Even recently? I fixed a lot of stuff with time positioning in replay a while back. I also think there may be some issue with Mavlink 1 versus Mavlink 2 streams, but not surel. Hence I need a log which shows the problem. Just SITL logs work fine.

Antiheavy, 2019-05-04:
Yes even recently. We are still using mavlink 1. Maybe it is time to change to 2. I’ll find a log and attach it here.

Antiheavy, 2019-05-05:
[2019-04-29 15-15-17.zip](https://github.com/mavlink/qgroundcontrol/files/3145719/2019-04-29.15-15-17.zip)

short example log attached.  scanning and scrubbing is sort of weird even with the newest Daily build.

DonLakeFlyer, 2019-06-11:
This should be fixed now in daily builds

closed 2019-06-11 by DonLakeFlyer