mavlink/qgroundcontrol #7416: (Android)Initial Time Value Issue in Flight Plan Tab

AGurel, 2019-05-04:
For android devices, in Flight Plan tab, initial time value is 01:00:00. However in other devices no problem.

I solve it as below;

In src/PlanView/PlanToolBar.qml file, line 78, i added the bold section. In debug mode, i saw that _missionTime's value is NaN, but after click Flight Plan tab the _missionTime's value is 0.

function getMissionTime() {
        if(isNaN(_missionTime) **|| _missionTime == 0**) {
            return "00:00:00"
        }
        var t = new Date(0, 0, 0, 0, 0, Number(_missionTime))
        return Qt.formatTime(t, 'hh:mm:ss')
    }

[image]
[image]

DonLakeFlyer, 2020-11-21:
Sorry I lost track of this bug. I installed both stable and daily and they both worked fine. From the screenshots It looks like you are running an old build.

DonLakeFlyer, 2020-12-30:
I'll reopen if you can make this happen on a new daily.

DonLakeFlyer, 2021-01-06:
@AGurel Can you take a screenshot with the Vehicle Info section of Mission Start expanded?

AGurel, 2021-01-07:
> @AGurel Can you take a screenshot with the Vehicle Info section of Mission Start expanded?

ok. you can see it below.

[image]

closed 2021-01-07 by DonLakeFlyer