mavlink/qgroundcontrol #7425: Import of mission files from OSX to Android does not work

jstrebel, 2019-05-08:
On both sides I have 3.5.2. I get a error message on the Android QGC: Error loading Plan file (/storage/emulated/0/QGroundControl/Missions/rene_u_urs.plan). Mission: The following required keys are missing: ValueSetIsDIstance,ImageDensity,FrontalOverlap,SideOverlap
I added two files. The file rene... was created with 3.5.0 and the file 7-mai.... created with the 3.5.3 on OSX
[rene_u_urs.plan.zip](https://github.com/mavlink/qgroundcontrol/files/3158863/rene_u_urs.plan.zip)
[7-mai_riedhof-mission.plan.zip](https://github.com/mavlink/qgroundcontrol/files/3158864/7-mai_riedhof-mission.plan.zip)

DonLakeFlyer, 2019-05-14:
Are you running different translated versions of QGC on each device? Like German on one but English on the other for example?

jstrebel, 2019-05-16:
Don, on my MAC OSX the Menue Texts are in German but the voice messages are in English.
On the Android, all are in English.

jstrebel, 2019-05-16:
Is there a way to set the Android device same as the OSX?

DonLakeFlyer, 2019-05-16:
I assume so, but not sure how you do that. There is a workaround in that you can modify your Plan files manually to make them work. If you open a Plan file in a text editor you will find a section like this:
```
                    "CameraCalc": {
                        "AdjustedFootprintFrontal": 15.424999999999997,
                        "AdjustedFootprintSide": 20.566666666666663,
                        "CameraName": "Manuell (keine Kamera Angaben)",
                        "DistanceToSurface": 110,
                        "DistanceToSurfaceRelative": true,
                        "version": 1
                    },
```

The problem happens due to the fact that the "CameraName" field is in German. If you want to open this plan file on an English system change it to look like this and it will load:
```
                        "CameraName": "Manual (no camera specs)",
```

jstrebel, 2019-05-17:
Don, Thank you the work around works. Jakob

DonLakeFlyer, 2020-11-20:
This has been fixed. But requires resaving plan files using a new build and then you can load using newer build even if language is different.

closed 2020-11-20 by DonLakeFlyer