mavlink/qgroundcontrol #7427: Rally point upload/download sometimes "not supported"

rmackay9, 2019-05-09:
ArduPilot is adding support for the MISSION_ITEM protocol (PR is here:https://github.com/ArduPilot/ardupilot/pull/10409) but during testing with QGC we are finding that, for whatever reason, QGC is often (but not always) showing "This vehicle does not support Rally Points". 

Here it is not working:
[image]

Here it is working moments later:
[image]

rmackay9, 2019-05-09:
I wonder if perhaps QGC is not requesting the AUTOPILOT_VERSION using MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES because I'm seeing this odd behaviour:

1. Start board and QGC and connect - it will not be possible to upload rally point
2. Shutdown QGC, connect with MP
3. Disconnect MP, re-connect with QGC - it will be possible to upload rally points

peterbarker, 2019-05-09:
The first two lines of output here are my debug:

```
##### Got autopilotversion!
setcapa(48079)
VehicleLog: "Vehicle does support MAVLink 2 but the link does not allow for it."
VehicleLog: "Vehicle supports Mavlink 2.0"
VehicleLog: "Vehicle supports MISSION_ITEM_INT"
VehicleLog: "Vehicle supports MISSION_COMMAND_INT"
VehicleLog: "Vehicle does not support GeoFence"
VehicleLog: "Vehicle supports RallyPoints"
VehicleLog: Delaying _startPlanRequest due to parameters not ready
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _sendMessageMultipleNext: 66
VehicleLog: _startPlanRequest
qrc:/qml/APMSafetyComponentSummary.qml:54: TypeError: Cannot read property 'enumStringValue' of null
qrc:/qml/APMSafetyComponentSummary.qml:70: TypeError: Cannot read property 'enumStringValue' of null
qrc:/qml/APMSafetyComponentSummary.qml:136: TypeError: Cannot read property 'value' of null
VehicleLog: _missionLoadComplete GeoFence not supported skipping
VehicleLog: _missionLoadComplete Rally Points not supported skipping
VehicleLog: _missionLoadComplete _initialPlanRequestComplete = true
MissionControllerLog: _newMissionItemsAvailableFromVehicle flyView:count true 0
MissionControllerLog: loading from vehicle: count 0
MissionControllerLog: _addMissionSettings addToCenter false
MissionControllerLog: MissionController::_scanForAdditionalSettings count:scanIndex 1 0
```

DonLakeFlyer, 2019-05-10:
@peterbarker I probably won't be able to get to this till the weekend. If you can repro easily did you double check AUTOPILOT_VERSION capability bits in mavlink inspector to make sure they are set correctly all the time?

DonLakeFlyer, 2019-05-10:
@peterbarker So even in the cases where rally points ui say not support your debug output says "it is supported"?

DonLakeFlyer, 2019-05-10:
@peterbarker Sorry, duh, I'm a moron I see it now. Trying to get some comments in fast before I head out.

peterbarker, 2019-05-10:
On Fri, 10 May 2019, Don Gagne wrote:

> @peterbarker Sorry, duh, I'm a moron I see it now. Trying to get some comments in fast before I head out.

Thanks for looking at this.

I believe it comes down to which version of mavlink QGC thinks the 
autopilot is capable / doing - I think the capa bits are just fine.

Note that ArduPilot doesn't support the "get protocol version" message 
(and it wasn't met with a vast amount of enthusiasm when it was brought up 
at a devcall), so QGC is relying on other means to set the mavlink version 
to something greater than 1 - and I believe that's a race condition so 
things *sometimes* work :-)

Reproduction is easy - plug an autopilot running this new branch into the 
USB port and fiddle.

DonLakeFlyer, 2019-05-11:
The QGC code checks for the capability bit as well as mavlink 2 on the link. So there is a bug in the mavlink 2 detection which should still work without the protocol version stuff.

DonLakeFlyer, 2019-05-11:
@peterbarker I built your branch and tried it in SITL but it's not reporting rally points as a set capability bit. Do I need to do something else to turn it on? Maybe I have something screwed up in my SITL build, not sure.

peterbarker, 2019-05-12:
On Sat, 11 May 2019, Don Gagne wrote:

> @peterbarker I built your branch and tried it in SITL but it's not reporting rally points as a set capability bit. Do I need to do something else to turn
> it on? Maybe I have something screwed up in my SITL build, not sure.

To be clear, you are expecting this bit, right?

https://github.com/ArduPilot/ardupilot/pull/10409/files#diff-4f15a04337259328467a92d8e35bab2fR4886

That bit did seem to be being set for me - and things have always worked 
for me in QGC run from SITL...

DonLakeFlyer, 2019-05-12:
My screwup with wrong PATH environment variable for SITL.

In the case where you see this happening do you get a "Communication Lost" while connecting which then goes away?

DonLakeFlyer, 2019-05-12:
@peterbarker Are you building QGC. If so can you try this. Remove this line and try again:
https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L2797

peterbarker, 2019-05-12:
On Sun, 12 May 2019, Don Gagne wrote:

> In the case where you see this happening do you get a "Communication Lost" while connecting which then goes away?

No, I've never seen that one.  But it could certainly happen with the way 
SITL's startup works (we reboot some of the vehicles to let certain 
parameters take effect).

peterbarker, 2019-05-13:
On Sun, 12 May 2019, Don Gagne wrote:

> @peterbarker Are you building QGC. If so can you try this. Remove this line and try again:
> https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L2797

I tried commenting that one out:

          } else if (!active && !_connectionLost) {
-            _updatePriorityLink(false /* updateActive */, false /* 
sendCommand */);
+            ::fprintf(stderr, "not updating PriotityLink");
+            // _updatePriorityLink(false /* updateActive */, false /* 
sendCommand */);

That codepath doesn't seem to be crossed - that fprintf is never emitted.

(apologies for the ham-fisted debug, it's just a reflex :-) )

DonLakeFlyer, 2019-05-13:
Ah, bummer. Still digging then. There is some sort of race condition I can't find yet.

DonLakeFlyer, 2019-05-14:
@peterbarker So I'm trying to build your branch so I can put it on a board. I tried:
```
./waf configure --board Pixhawk4
./waf copter
```

But I always get this error:
```
[6/8] Running uavcangen
[7/8] Creating build/Pixhawk4/ap_version.h
Waf: Leaving directory `/home/parallels/repos/peter/build/Pixhawk4'
Build failed
Traceback (most recent call last):
  File "/home/parallels/repos/peter/modules/waf/waflib/Task.py", line 338, in process
    ret = self.run()
  File "Tools/ardupilotwaf/uavcangen.py", line 24, in run
    input_dir = os.path.dirname(self.inputs[0].abspath())
IndexError: list index out of range
```

DonLakeFlyer, 2019-05-14:
Possibly instead of me fighting with that if you could give me a Pixhawk4 version of your code I can flash that would help as well.

DonLakeFlyer, 2019-05-17:
I believe the reason is because the link starts out a Mavlink 1. Then MP somehow magically switches the link to Mavlink 2. So when you connect again with QGC the link is running mavlink 2 and it works. 

If so, I don't understand how ArduPilot expects a ground station to decide to use Mavlink 2 and how that works with respect to dealing with radios which can't pass mavlink 2 traffic.

DonLakeFlyer, 2019-05-17:
Yup, I think what I said above is correct. If QGC always sends Mavlink 2 then ArduPilot will switch to Mavlink as well. So the reason this is not working is because the link stays at mavlink 1.

peterbarker, 2019-05-17:
On Fri, 17 May 2019, Don Gagne wrote:

> Yup, I think what I said above is correct. If QGC always sends Mavlink 2 then ArduPilot will switch to Mavlink as well. So the reason this is not working
> is because the link stays at mavlink 1.

Yep, that sounds correct.

We force mavlink2 if signing is enabled: 
https://github.com/ardupilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L162

Otherwise we continue to send mavlink1 on a port (even if it marked as 
mavlink2 using SRn_PROTOCOL!) until we see any mavlink2 traffic on that 
port: 
https://github.com/ardupilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L1446

So both ends are waiting for the other to talk mavlink2, is that correct?

Note that ArduPilot does indicate it is mavlink2 capable if the port is 
set to mavlink2: 
https://github.com/ardupilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L4638

Now I don't know ArduPilot's behaviour is specs-compliant here.  (anybody 
know where the spec lives? :-) )

Could QGC check the mavlink2 capabilities bit and switch to mavlink2 if it 
sees it?

peterbarker, 2019-05-17:
On Sat, 18 May 2019, Peter Barker wrote:

> Could QGC check the mavlink2 capabilities bit and switch to mavlink2 if it 
> sees it?

Oh, duh.  Sorry, short on coffee here.  I've already looked at this, and 
QGC *does* look for that bit.

ArduPilot sets the bit if the *link* is capable of mavlink2 - so on some 
links it will be true, others false.

QGC seems to be looking at the bit for vehicle compatability and then 
checking the link for protocol version compatability...

DonLakeFlyer, 2019-05-17:
> So both ends are waiting for the other to talk mavlink2, is that correct?

Basically. 

I found this today: https://github.com/mavlink/mavlink-devguide/blob/master/kr/guide/mavlink_version.md which does explain how QGC deals with the negotiation. At least I think it is correct. I'm still trying to figure out how it all works myself since it's not code I added.

How does ArduPilot expect a ground station to work around the problem of telemetry radios which don't support Mavlink 2. The capability bit can say the firmware supports mav 2, but it still doesn't quite mean you can use it. Does the user just have to deal with that by setting SERIAL0_PROTOCOL manually to Mavlink 1 if they have that setup?

DonLakeFlyer, 2019-05-17:
> Does the user just have to deal with that by setting SERIAL0_PROTOCOL manually to Mavlink 1 if they have that setup?

If that is the case, then QGC could just rely on the capability bit and if it sees the bit set then it starts mavlink 2 from it's side. But for ArduPilot vehicles only.

DonLakeFlyer, 2019-05-17:
Also FYI I can reproduce the problem using master ArduCopter firmware so I can just flash that and don't need to build my own using waf. That is how I've been testing.

peterbarker, 2019-05-17:
On Fri, 17 May 2019, Don Gagne wrote:

> I found this today: https://github.com/mavlink/mavlink-devguide/blob/master/kr/guide/mavlink_version.md which does explain how QGC deals with the
> negotiation. At least I think it is correct. I'm still trying to figure out how it all works myself since it's not code I added.

I think the critical thing there might be - is QGC sending that 
PROTOCOL_REQUEST as a mavlink2 packet?  That - I think - should be 
sufficient for ArduPilot for switch to doing mavlink2.

> How does ArduPilot expect a ground station to work around the problem of telemetry radios which don't support Mavlink 2. The capability bit can say the
> firmware supports mav 2, but it still doesn't [...]

Welll... SERIAL0 is *magic*, so we should probably not talk about that one 
:-)

ArduPilot currently defaults everything to mavlink1 - except SERIAL0 - so 
a user has to explicitly try out mavlink2 by setting (e.g.) 
SERIAL1_PROTOCOL to 2.

peterbarker, 2019-05-17:
On Fri, 17 May 2019, Don Gagne wrote:

> I found this today: https://github.com/mavlink/mavlink-devguide/blob/master/kr/guide/mavlink_version.md which does explain how QGC deals with the
> negotiation. At least I think it is correct. I'm still trying to figure out how it all works myself since it's not code I added.

The English (the link you pointed at seems to be a rather Englishified 
version of Korean...) version may differ: 
https://github.com/mavlink/mavlink-devguide/blob/master/en/guide/mavlink_version.md

Note in particular:

----
Vehicles should set the MAV_PROTOCOL_CAPABILITY_MAVLINK2 capability flag 
in the AUTOPILOT_VERSION message if MAVLink 2 is available on a link. This 
should be set in the case where the link is currently sending MAVLink 1 
packets but MAVLink 2 packets will be accepted and will cause a switch to 
MAVLink 2.
----

So I *think* we're compliant with these guidelines (presumably made by 
someone who has thought about this a bit more thoroughly than I have!). 
In particular, there do appear to be two suggested ways on that page of 
"negotiating" mavlink2 - the protocol-request thing and autopilot-version 
thing.

DonLakeFlyer, 2019-05-17:
> Vehicles should set the MAV_PROTOCOL_CAPABILITY_MAVLINK2 capability flag in the AUTOPILOT_VERSION message if MAVLink 2 is available on a link. This should be set in the case where the link is [...]

Interesting because that isn't quite how QGC is currently coded. Let me look through that doc some more to see if I can make sense of it.

Yeah, my screwup on SERIAL0 not a telem port.

DonLakeFlyer, 2019-05-17:
> is QGC sending that PROTOCOL_REQUEST as a mavlink2 packet?

Nope

DonLakeFlyer, 2019-05-17:
> Which is odd, because the spec says it should be mavlink2.

No wrong. The QGC sends mavlink 1 PROTOCOL_REQUEST, but the vehicle should then respond in mavlink 2.

peterbarker, 2019-05-18:
On Fri, 17 May 2019, Don Gagne wrote:

>       Which is odd, because the spec says it should be mavlink2.
> No wrong. The QGC sends mavlink 1 PROTOCOL_REQUEST, but the vehicle should then respond in mavlink 2.

Yeah, I think you're right there.  We should probably find some words to 
clarify these documents.

So the current work-around is to force qgc to do mavlink2.

Options to fix include:

  - ArduPilot to support PROTOCOL_VERSION
   - trivial technically AFAICS, not sure about acceptance from other devs

  - OR QGC to switch to sending mavlink2 based on the following two lines:

- Vehicles should set the MAV_PROTOCOL_CAPABILITY_MAVLINK2 capability flag 
in the AUTOPILOT_VERSION message if MAVLink 2 is available on a link. This 
should be set in the case where the link is currently sending MAVLink 1 
packets but MAVLink 2 packets will be accepted and will cause a switch to 
MAVLink 2.
- GCS implementations can choose to either automatically switch to MAVLink 
2 where available or to have a configuration option for MAVLink 2.

Which basically comes down to QGC trusting the definition of the 
capability bit, I think.

I've just noticed that RAW we should turn that bit off if we're currently 
doing mavlink2 on a link :-)

DonLakeFlyer, 2019-05-18:
@peterbarker Can you try https://github.com/DonLakeFlyer/qgroundcontrol/tree/MavlinkNegotiate?

The whole thing was quite a mess but I think at a minimum I have single vehicle negotation working correctly now. The new code should switch the link to mavlink 2 if the capability is set and the vehicle does not support PROTOCOL_REQUEST.

I believe multi-vehicle multi-link negotation is broken and likely was broken before anyway if you happened to have vehicles or links with differing mavlink 2 capabilities.

Also I'm heading out of town tomorrow and this change is pretty major so I'm not going to be able to check it in before I leave. It needs more bake time and testing to make sure I haven't royally screwed something up and leave a trashes daily build.

Also 2: I noticed something strange when I switched SERIAL0_PROTOCOL to Mavlink 1. I no longer could connect to the board. I tried MP as well and it couldn't connect either. Not sure if that has something to do with my setup but it seemed kind of real after trying various things to get it to connect. I was running maste Copter ChibiOS firmware.

peterbarker, 2019-05-20:
On Sat, 18 May 2019, Don Gagne wrote:

> Also 2: I noticed something strange when I switched SERIAL0_PROTOCOL to Mavlink 1. I no longer could connect to the board. I tried MP as well and it
> couldn't connect either. Not sure if that has [...]

Just tested here with QGC and MAVProxy and they connect to a 
near-enough-to-master ArduPlane without issue where SIERAL0_PROTOCOL is 
set to 1.

hamishwillee, 2019-05-20:
FYI, the rendered version of the versioning doc is here: https://mavlink.io/en/guide/mavlink_version.html

If the result of the discussion was a flaw in these docs - something missing, unclear or incorrect, please can you post a defect against the page?

anandbaskaran, 2019-05-31:
> I believe the reason is because the link starts out a Mavlink 1. Then MP somehow magically switches the link to Mavlink 2. So when you connect again with QGC the link is running mavlink 2 and it works.
> 

This discussion means that Mavlink 2 is required for the rally points and geofence. Am I right?

peterbarker, 2019-06-02:
On Fri, 31 May 2019, Anand wrote:

> I believe the reason is because the link starts out a Mavlink 1. Then MP somehow magically switches the link to Mavlink 2. So when you connect
>       again with QGC the link is running mavlink 2 and [...]

That's correct - the mission type field is an extension.

DonLakeFlyer, 2019-06-04:
@peterbarker Have you had a chance to try my branch?

DonLakeFlyer, 2019-06-17:
Fixed

closed 2019-06-17 by DonLakeFlyer

DonLakeFlyer, 2019-08-14:
FYI: To folks testing this. The change is only in Daily builds. It's a bit too involved to retrofit to Stable.

rmackay9, 2019-10-26:
I think this issue still persists.  I loaded the latest daily build (just today) and found that I was not able to upload fences to my multicopter running Copter-4.0.0-rc1 unless I connected to MP once first.