mavlink/qgroundcontrol #7430: Param Write for Mavlink Components wrong

olliw42, 2019-05-11:
## Expected Behavior
when writing a parameter of a component, the new value should be converted into the MAVLink PARAM_SET message using the mavlink_param_union_t structure, which results in a typecast, and not just set be coding param.value = newvalue, which results in a conversion to float.

For instance, a parameter of type UINT8 and of new value 71 should lead to bytes '0x47 0x00 0x00 0x00' in the MAVLink stream, and NOT '0x00 0x00 0x8E 0x42' (which corresponds to the byte stream of a float of value 71,  0x42 0x8E 0x00 0x00 = 71.0f)

## Current Behavior
Currently the behavior corresponds to param.value = newvalue, i.e. a conversion to float.

## Steps to Reproduce:
connect a component such as a STorM32 gimbal controller to a flight controller, call QGC, write a parameter to the component and spy the actual byte stream coming out of QGC or which is reaching the component.

## System Information
- Operating System: Windows 7
- QGC Version: 3.5.0
- QGC build: stable
- Flight Controller: PixRacer, but irrelevant IMHO
- Autopilot (with version): Ardupilot Copter 3.6.7, but irrelevant IMHO

## Detailed Description
I use a STorM32 gimbal controller connected to a pixracer, which in turn is connected to the PC. When using QGC writing the component's parameters produces wrong results, while when I use MissionPlanner, it all works perfectly fine.

I have created a means which allows me to spy the actual byte stream which is received by the STorM32 gimbal controller, which unambiguoulsy shows the above wrong behavior. Please see the next attached screenshots

## Log Files and Screenshots
screenshot of the byte stream of a PARAM_SET when using QGC:
[image]

screenshot of the byte stream of a PARAM_SET when using MissionPlanner:
[image]

In both pictures the long message in about the middle of the screen is the PARAM_SET message as it is send to the STorM32 controller. It is clearly seen that QGC sends the bytes '0x00 0x00 0x8E 0x42' instead of '0x47 0x00 0x00 0x00' in the param value section.

DonLakeFlyer, 2019-05-11:
> * Ardupilot Copter 3.6.7, but irrelevant IMHO

This is extremely relevant in this case. ArduPilot does not support mavlink_param_union_t. All params are sent over the wire as float no matter what type they are. This is all working correctly with respect to how ArduPilot works.

olliw42, 2019-05-11:
no, you are wrong
it has nothing to do with ArduPilot, it all has ONLY to do with QGC!
ArduPilot is NOT modifying the content of these messages at all, it is only routing it from QGC to the component and back from the componet to QGC.

frankly, I think you just want to scrub it off quickly following a reflex, while thinking logically would be more appropriate 

miracously it's working correctly with MP but not QGC, immediately disproving your argument, and miracously QGC displays the correct value when it reads it but fails when it writes it, further disproving your argument

pl, make an attempt to analyze the evidence logically

DonLakeFlyer, 2019-05-11:
Right, sorry this is a component not the main autopilot.

> frankly, I think it would be useful if you would not just follow quick reflexes

When you are the sole person who looks at incoming issues and 99% of the ones entered are not real issues that can be a difficult thing to do. Sorry about that.

olliw42, 2019-05-11:
well, I perfectly understand what you're saying
but, you know, the basic argument that ArduPilot wouldn't consider the union has been true several years ago but isn't anymore since a long while ... ;)

I didn't expected this to be reopenable again, so I've created a new issue. I close this here.

:)

closed 2019-05-11 by olliw42

hamishwillee, 2019-05-12:
> but, you know, the basic argument that ArduPilot wouldn't consider the union has been true several years ago but isn't anymore since a long while ... ;)

FYI @olliw42 Incorrect. Fortunately Tridge is working on a new param protocol that uses a completely different approach. I think it will be sufficiently good improvement that it will go into common and supersede the current protocol.