mavlink/qgroundcontrol #7431: Param Write for Mavlink Components wrong

olliw42, 2019-05-11:
In contrast to the claim in #7430 the following has nothing to do with ArduPilot, since ArduPilot is not modifying messages which it routes from the GCS to the component and vice versa (and the argument given in #7430 is wrong anyhow). I thus reopen.

## Expected Behavior
when writing a parameter of a component, the new value should be converted into the MAVLink PARAM_SET message using the mavlink_param_union_t structure, which results in a typecast, and not just set be coding param.value = newvalue, which results in a conversion to float.

For instance, a parameter of type UINT8 and of new value 71 should lead to bytes '0x47 0x00 0x00 0x00' in the MAVLink stream, and NOT '0x00 0x00 0x8E 0x42' (which corresponds to the byte stream of a float of value 71,  0x42 0x8E 0x00 0x00 = 71.0f)

## Current Behavior
Currently the behavior corresponds to param.value = newvalue, i.e. a conversion to float.

## Steps to Reproduce:
connect a component such as a STorM32 gimbal controller to a flight controller, call QGC, write a parameter to the component and spy the actual byte stream coming out of QGC or which is reaching the component.

## System Information
- Operating System: Windows 7
- QGC Version: 3.5.0, EDIT: just confirmed that 3.5.2 shows the same behavior
- QGC build: stable
- Flight Controller: PixRacer, but irrelevant IMHO
- Autopilot (with version): Ardupilot Copter 3.6.7, but irrelevant IMHO

## Detailed Description
I use a STorM32 gimbal controller connected to a pixracer, which in turn is connected to the PC. When using QGC writing the component's parameters produces wrong results, while when I use MissionPlanner, it all works perfectly fine.

I have created a means which allows me to spy the actual byte stream which is received by the STorM32 gimbal controller, which unambiguoulsy shows the above wrong behavior. Please see the next attached screenshots

## Log Files and Screenshots
screenshot of the byte stream of a PARAM_SET when using QGC:
[image]

screenshot of the byte stream of a PARAM_SET when using MissionPlanner:
[image]

In both pictures the long message in about the middle of the screen is the PARAM_SET message as it is send to the STorM32 controller. It is clearly seen that QGC sends the bytes '0x00 0x00 0x8E 0x42' instead of '0x47 0x00 0x00 0x00' in the param value section.

DonLakeFlyer, 2019-05-11:
Why another one? #7430 is fine.

DonLakeFlyer, 2019-05-11:
Can you attach a telemetry log for a connect to this vehicle which has the parameter download in it? I want to double check what it looks like with respect to component and system id for the gimbal to make sure I get the case correct.

olliw42, 2019-05-11:
you mean a tlog in the QGroundControl/Telemetry folder?
I was actually trying to look at this, but I could not find a log in there which would have a time stamp which remotely would have matched to when I did this  
do I have to actively save the log?
some brief instructions could be helpful, I'm unfortunately not yet very familiar with QGC

DonLakeFlyer, 2019-05-11:
> you mean a tlog in the QGroundControl/Telemetry folder?

Yup. You can always just empty out the folder. The telemetry log by default will only write out if you arm the vehicle. You can change it to write out without arming if you need to from General settings.

olliw42, 2019-05-11:
many thx

here we are: 
[2019-05-11 23-39-32.zip](https://github.com/mavlink/qgroundcontrol/files/3169495/2019-05-11.23-39-32.zip)

what I did:
I connected to the vehicle
I when went to the component's 154 parameters, and did a refresh
I when entered the value 71 into the CAN_UCNODEID field (I took this field since it's noncritical for the gimbal operation)

you should be able to see the above: i.e. in the read it correctly gets a value of 72, but in the write of the 71 it does it as float

note: there is also a 102 camera component, doing the camera magic

DonLakeFlyer, 2019-05-12:
I'm still waiting for clarification as to how to handle this from ArduPilot folks. So far not clear at all.

olliw42, 2019-05-13:
I've seen the communication at gitter. It's indeed a confusing response, and as you say there it doesn't look like a workable architecture. I feel somewhat sorry for having poked into this.

what confuses me generally is that the id appears to be taken as primary info of the type of component, and that the type and/or autopilot fields of the heartbeat seem to have become superfluous. Otherwise I would have thought that a component should use MAV_AUTOPILOT_INVALID, which would differentiate it unambiguously from an autopilot. And MAV_AUTOPILOT_ARDUPILOTMEGA would identify it unambiguously as ArduPilot. But I understand that I may miss the point.

DonLakeFlyer, 2019-05-14:
I'm still trying to figure out how this works fine with Mission Planner so I can duplicate that.

DonLakeFlyer, 2019-05-14:
> Otherwise I would have thought that a component should use MAV_AUTOPILOT_INVALID, which would differentiate it unambiguously from an autopilot.

That is apparently the key piece of information you can trigger on to sort of do the right thing.

DonLakeFlyer, 2019-05-14:
@olliw42 Can you try tomorrows daily build and see if it's fixed?

olliw42, 2019-05-15:
sure
could I install the latest this evening or would I better to wait until tomorrow?
(I ask because yesterday mid-day I installed the "daily" build using the Win installer, but the QGC title showed something of more like 2 days ago)

are there any other situations I could/should test? I mean, I can easily change the sysId, compId, the type field of the component, and such, and test if the behavior changes in the expected way.

I briefly looked at your code changes, and I can't say I really understand them. I note however that there is a test for `(message->compid == MAV_COMP_ID_AUTOPILOT1)` when determining the flying state,
https://github.com/DonLakeFlyer/qgroundcontrol/blob/83b500d0b126a8296301ba8ca46fafb537b64d28/src/FirmwarePlugin/APM/APMFirmwarePlugin.cc#L452
while for the map the test is `heartbeat.autopilot == MAV_AUTOPILOT_ARDUPILOTMEGA` as said in the PR
https://github.com/DonLakeFlyer/qgroundcontrol/blob/83b500d0b126a8296301ba8ca46fafb537b64d28/src/FirmwarePlugin/APM/APMFirmwarePlugin.cc#L467
Isn't this inconsistent, i.e. a bug? I mean, if it's `autopilot == MAV_AUTOPILOT_ARDUPILOTMEGA` which is used now as key for identifying an ArduPilot shouldn't it then also be used for correcting the flying state of an ArduPilot vehicle? I.e. `if ((autopilot == MAV_AUTOPILOT_ARDUPILOTMEGA) && vehicle->armed())` in line 454?
If it's nonsense what I'm saying pl just ignore.

DonLakeFlyer, 2019-05-15:
> I mean, if it's `autopilot == MAV_AUTOPILOT_ARDUPILOTMEGA` which is used now as key for identifying an ArduPilot shouldn't it then also be used for correcting the flying state of an ArduPilot vehicle?

No. The correct flying state only comes from the main autopilot firmware code which is the MAV_COMP_ID_AUTOPILOT1 component. The old code was actually incorrect in that it would look at that information coming from any component which would cause the vehicle state to be set incorrectly.

DonLakeFlyer, 2019-05-15:
The new build should be there now

DonLakeFlyer, 2019-05-15:
> If it's nonsense what I'm saying pl just ignore.

Nope. It's a good question and good to double-check.

olliw42, 2019-05-15:
>No. The correct flying state only comes from the main autopilot firmware code which is the MAV_COMP_ID_AUTOPILOT1 component. The old code was actually incorrect in that it would look at that information coming from any component which would cause the vehicle state to be set incorrectly.

I might have not expressed it well. I wasn't trying to say that this check should be removed or modified, but I would think that the following if check inside should be extended. The comment appears to say that this part is ardupilot specific. Maybe I write it out in more detail what I mean:
```
    if (message->compid == MAV_COMP_ID_AUTOPILOT1) {
        // We pull Vehicle::flying state from HEARTBEAT on ArduPilot. This is a firmware specific test.
        if ((heartbeat.autopilot == MAV_AUTOPILOT_ARDUPILOTMEGA) && vehicle->armed()) {
            flying = heartbeat.system_status == MAV_STATE_ACTIVE;
            if (!flying && vehicle->flying()) {
                // If we were previously flying, and we go into critical or emergency assume we are still flying
                flying = heartbeat.system_status == MAV_STATE_CRITICAL || heartbeat.system_status == MAV_STATE_EMERGENCY;
            }
        }
        vehicle->_setFlying(flying);
    }
```
The difference is in the 2nd if.

DonLakeFlyer, 2019-05-15:
Ok, I see. That extra check would always be true. If you had the combination of MAV_COMP_ID_AUTOPILOT1 and heartbeat.autopilot != MAV_AUTOPILOT_ARDUPILOTMEGA that would not be an ArduPilot vehicle. Hence you would not be in this code.

olliw42, 2019-05-15:
>That extra check would always be true. 

I don't understand this statement, and in fact find it wrong

if it would be a PX4 then the extra check would not be true, and without it the code inside the 2nd if would be executed even though it's not an ArduPilot in contrast to what the comment implies, namely that it should be executed only for an ArduPilot

>If you had the combination of MAV_COMP_ID_AUTOPILOT1 and heartbeat.autopilot != MAV_AUTOPILOT_ARDUPILOTMEGA that would not be an ArduPilot vehicle. Hence you would not be in this code.

with how it is currently you would be in that code part for this combination, even though one shouldn't according to the comment

what am I missing?

DonLakeFlyer, 2019-05-15:
> if it would be a PX4 then the extra check would not be true

If it was possible to end up in any of this code when the vehicle was running PX4 firmware then that statement would be true. But this is the ArduPilot firmware plugin which is only called if the Vehicle is already determined to be running ArduPilot firmware. Other firmware types would be directed to their own respective FirmwarePlugin code.

olliw42, 2019-05-15:
well, then it's vice versa, then the ` if (message->compid == MAV_COMP_ID_AUTOPILOT1)` is obsolete, because if that function is called it has been already determined that it's an ardupilot, which implies that this condition must have been checked already in the outside of the function

DonLakeFlyer, 2019-05-15:
The fact this code path is only taken for ArduPilot vehicles is unrelated to the fact that a vehicle can have multiple components on it with different ids. One of which is the main autopilot code (MAV_COMP_ID_AUTOPILOT1) which is the only component which can be used to determine flying state. Other components can use that state for whatever they want.

olliw42, 2019-05-15:
just installed the latest build, and played around a bit with it, and the parameter write appears to be all good now! At least I couldn't yet observe any misbehavior! GREAT. Thx so much.

There is however, I believe, another slight issue with the parameter write: 
The changes are not been made permanent. This is so because a change to a parameter is not directly made permanent in the STorM32 I use here as component, but requires an additional store step. This in turn is so in order to avoid that the FLASH (which is used to emulate an EEPROM) wears out (note: the FLASH on micro-controllers such as ATMEGA and STM32 and others can be erased only a relatively low number of times). You will find that this "write and store only if explicitly asked for" approach is a very typical approach used on quite many projects, and thus not really an odd feature of only the STorM32, but potentially of many components.

Mission Planner handles this by sending out a MAV_CMD_PREFLIGHT_STORAGE (with storage = 1) 10 seconds after the last parameter change operation (Michael actually added this also on my request few years back, the git issue conversation should be still retrievable, for a more complete discussion on the rational). This approach has turned out to work very well.

So, I kindly would ask you to also add this mechanism.

olliw42, 2019-05-15:
> Other components can use that state for whatever they want.

but this other component wouldn't be an ArduPilot by definition and thus the whole function wouldn't ever be called according to your statement. Ergo the if is pointless. I'm just logically combining the available information :).

Anyways, I might not see the point and I of course totally trust you that you do the right thing, so I surrender and guess to not continue discussing this point. I just wish to note however that my warning bells have triggered as regards to how the handling of autopilot and component identification is done. It doesn't appear to follow a logically consistent pattern across all autopilots and components (we also had this camera identification e.g. recently, which is strikingly different in logic). In my experience this rather creates issues than solves them. That's all. :)
As is, it is certainly working great for me.

DonLakeFlyer, 2019-05-15:
You're confusing the ArduPilot component mavlink strangeness with the internal QGC architecture. QGC firmware type for plugin routing is determined at a whole vehicle level, not at a component level. A vehicle can have multiple components associated with it. All component mavlink traffic within that same vehicle is routed to the same plugin.

DonLakeFlyer, 2019-05-15:
> It doesn't appear to follow a logically consistent pattern across all autopilots and components

It would if everyone followed current mavlink spec. Which is not the case. So ground stations which try to support multiple firmware types are stuck in a self-imposed nightmare.

olliw42, 2019-05-15:
> QGC firmware type for plugin routing is determined at a whole vehicle level, not at a component level. ... All component mavlink traffic within that same vehicle is routed to the same plugin.

ah ... so, the camera or gimbal or whatever stuff would go through different codes in QGC dependent on the main vehicle ... I certainly didn't realized or imagined that. This was helpful. Thx.

olliw42, 2019-05-16:
@DonLakeFlyer 

as regards the "slight issue with the parameter write" which I reported 5 posts up, I reviewed the situation and came to the following conclusion:

The procedure used in Mission Planner to send out a `MAV_CMD_PREFLIGHT_STORAGE` 10 seconds after the last parameter change was largely based on the rational that the MAVLink spec for the `PARAM_SET` message at this time was explicitly against writing it immediately to permanent storage, see original conversation ArduPilot/MissionPlanner#882. 

However, I now realize that the MAVLink spec for `PARAM_SET` has changed and now explicitly states "(write new value to permanent storage)" (https://mavlink.io/en/messages/common.html#PARAM_SET).

This is going to cause quite some grievance now, but it's not QGC's fault. QGC is doing the right thing, and I take back my request formulated in the above. Since the main topic of this issue has been nicely resolved, I'm also closing this issue.

Many thx, sir.

DonLakeFlyer, 2019-05-16:
> However, I now realize that the MAVLink spec for `PARAM_SET` has changed and now explicitly states "(write new value to permanent storage)"

Both PX4 and ArduPilot firmwares changed away from requiring the PREFLIGHT_STORAGE thing years ago. So at some point this was the case, but no longer and it look like the spec may have changed to reflect what they were doing.

olliw42, 2019-05-16:
thx for clarification. It makes sense for the flight controllers. 

I also could "easily" handle this in the STorM32 by simply doing this 10 second thing internally, i.e. to store the parameters permanently after a timeout of 10 secs after the last PARAM_SET. Easy enough.

However: I would not know what to do with the PREFLIGHT_STORAGE coming from MissionPlanner. Since one can't reliably determine who is sending it, I can't just blend it out. I could remove this message altogether, which would be a functional loss however. Ideally, MissionPlanner would undo this things, but it's probably not clear how that would impact all other components which might be out there and started relying on it. A double timeout thing, i.e. store 10+X secs after a parameter change, but don't do it if the last store was X secs ago, would blend MPs store out, but might not be totally reliable in all situations. 

Fact is, QGC and MP are currently very different in this aspect, and it's not so obvious how a component could cope with that. Very unfortunate situation :)

DonLakeFlyer, 2019-05-16:
The functional loss of not be able to make temporary changes to params which don't stick across a reboot doesn't seem super valuable.

Seems like it would be way better to track if permanent storage update is required internally with a dirty bit. If so, do it on a delay. If you get a PREFLIGHT_STORAGE command when it is not needed just ignore it since it's a no-op. Given the current mavlink spec you can't trust that anybody new is going to be sending that command any more. Especially since that spec change happened years ago. That way moving forward you are covered.

Which still doesn't fix the problem of your gimbals out there with existing firmware which won't automatically store param changes. Which means that QGC should really do some sort of delayed PREFLIGHT_STORAGE on non-autopilot components.

DonLakeFlyer, 2019-05-16:
My recommondation would be to do both. QGC does PREFLIGHT_STORAGE thing for additional components. That gets pulled out some time down the road say in a year or so. Gimbal firmware updates to manage storage by itself without needing PREFLIGHT_STORAGE which future proofs you to the current spec.

olliw42, 2019-05-16:
MANY THX for being so forthcoming, I really appreciate this a lot

I'd however like to recommend you to NOT add the PREFLIGHT_STORAGE thing :)

First, as you say, it would (and should) be temporary, which in turn means that a solution must eventually be established anyways. So why not now.
Second, it's really not like that the number of STorM32 users is so massive that it would warrant such efforts on your side. In fact, since the DJI P4 the number of DIY gimbal builders and therewith STorM32 users has massively dropped. And of those most still just use the cheap I2C board for which I stopped firmware support ca 3 years ago. So, the number of STorM32 users who could do a firmware upgrade I assume is very small (I have absolutely no idea of realistic values, but it's certainly small on any scale relevant to you!). And those STorM32 users who do use upgradable setups I believe are always quick with upgrading.

So, if you would do that for only the reason to support STorM32 users, then I VERY much appreciate your willingness, but you definitely shouldn't do it for this reason. It really unecesarily would eat up your time.

DonLakeFlyer, 2019-05-16:
> So, if you would do that for only the reason to support STorM32 users

That was my main reason. I'll leave things alone then.

closed 2019-05-16 by DonLakeFlyer

hamishwillee, 2019-05-17:
> Both PX4 and ArduPilot firmwares changed away from requiring the PREFLIGHT_STORAGE thing years ago. So at some point this was the case, but no longer and it look like the spec may have changed to reflect what they were doing.

Just FYI

Yes, I did this sometime last year. There is no way to specify that params should be in volatile or permanent storage, and in fact everyone was writing to permanent storage. The dev call signed off on this.

If I could go back in time, I might add a message extension to cover the two options. That would still be possible if there was strong interest.

olliw42, 2019-05-17:
> everyone was writing to permanent storage. The dev call signed off on this.

I would then suspect that in this discussion the focus wasn't much on components but mostly on flight controllers !? 

It would actually be interesting to know how many of the existing MAVLink components are included in this "everyone". So, may I asked, is this based on data you guys have at hand, or for components would be more a guess?

> If I could go back in time, I might add a message extension to cover the two options. That would still be possible if there was strong interest.

As I just told DonLakeFlyer, you shouldn't consider doing such things if it's only for the STorM32, the STorM32 users are AFAIK and IMHO not very relevant to you. You might be more concerned about BASECam, Gremsy, ++++ users.

The problem is not the current spec, but that the spec has changed. Like for DO_MOUNT.  The "correct" solution would, like for DO_MOUNT, to ask MissionPlanner to undo the thing, with the argument that this is according to the spec. Component maintainers have then to deal with this.

The more fundamental problem is that the new spec doesn't reflect the micro-controller reality (that their FLASH has a rather limited erase number). 

And it might have now actually consequences which were not considered.

For instance, I just learned that there are strong reasons for a camera to have a CAM_MODE parameter. As much as I understand QGC modifies it by PARAM_SET (or PARAM_EXT_SET, respectively). So when one is using a camera definition file in order to make best use of this super cool camera thing, one ends up to kind of continuously erase the FLASH! With each change of the mode the FLASH ticker goes down! This will not only wear out FLASH quickly, reducing life time of the product, but erasing FLASH is a dammed slow operation too, taking several ms. So what this means is that each time the camera mode is changed the gimbal controller loop has to stop for a pretty long time ... twitch, twitch, twitch. What a nightmare. 
So, I have already decided that the STorM32's camera controller will have to break the spec and will NOT write the camera component parameters into permanent storage. Whatever confusion this will bring for the user.

So, if you were to ask me, I would undo this move and go back to the "old" spec, where PARAM_SET makes a non-permanent write, and the STORAGE command to make it permanent. That's the obvious scheme and there are reasons for why it has emerged over the years in so many projects. I would not fool around with new flags etc, since it makes things just more complicated and there is already a well established, proven and tried procedure for handling this micro-controller oddity, namely exactly this write & store. Why replacing a good working wheel by a complicated whatever.

However, I think that if I were you I probably wouldn't listen to me. :)

seemingly simple change with lots of implications :)

hamishwillee, 2019-05-17:
@olliw42 The simple answer is yes, we (or at least "I") did not think about components. The change seemed rational as the messages were clearly wrong for autopilots. It is indeed unfortunate that the behaviour did make sense for components. 

We should ensure that there is a mechanism for both cases. One way is to just change the docs to remove "to permanent storage" and assume the device decides what is appropriate. That is actually what happens.

The other is to provide an extension field to the param set that makes this explicit - ie with an enum - permanent, temporary, auto.

Thoughts?

olliw42, 2019-05-17:
> The change seemed rational as the messages were clearly wrong for autopilots.

I guess I wouldn't totally agree with that. I would differentiate devices which use an EEPROM or not. It's probably true that this once was a tautology, as autopilots used EEPROMs, but not anymore.

For instance, ArduPilot's recent move to Chibios opened the way for many of the cheap flight controllers which mostly don't have a EEPROM but use the FLASH for parameter storage. I would not be surprised at all if in 2-3 years "suddenly" many reports of suddenly dead FC would pile up ...

> One way is to just change the docs to remove "to permanent storage" and assume the device decides what is appropriate.

I would strongly advice against not clearly saying what the supposed behavior is ... if everyone is going to do their own thing there is zero chance to get some predictable behavior

> The other is to provide an extension field to the param set that makes this explicit - ie with an enum - permanent, temporary, auto.

for the same reasons there must not be an option auto, IMHO

I see it it this way: If it would be so easy for the device to decide what is appropriate, which here means to determine what the intention of the sender was, then the whole discussion would not exist. Sadly, only the sender has a real chance to know what the intention of sending it is, and any attempt of a receiver to figure out the senders intention necessarily will be guesswork. So, no auto, and precise specs, IMHO.

Also, I'm not convinced enums really help a lot: The FLASH on the micros can be erased only page wise, and not byte/word wise. Thus, storing a single new parameter requires one to erase a full page. So, if one would do two writes, and at each write a flag tells to store it permanently, one would end up erasing the page twice. Which is obviously not desirable. In order to avoid that, the device would thus again have to do some sort of timeout-ing, bit-ing, or whatever, to decimate the page erases, and little if not nothing would have been gained.
Pl also realize that all such schemes introduce some period of delay of the action, which might lead to all sorts of troubles by itself, so in any case one then needs to add another option PERMANENT_IMMEDIATELY to the enum.
I fail to see any good argument for why one would prefer such a complicated approach over the simple write&store. IMHO it is a no-brainer for a GCS to decide when to send a store message, and any autopilot is free to store immediately upon a PARAM_SET and ignore any store messages. So, it all just boils down to that a GCS doesn't want to send store messages.

So, I uphold my opinion that I would revert it to the old. It's the only scheme I can see which ends up in something clean and trouble-free. With any other approach known to me I just see lots of entry doors for lots of issues.

If reverting is not an option, which I think you guys may conclude, then the enum should have three options 

TEMPORARY, PERMANENT, PERMANENT_IMMEDIATELY

with TEMPORARY the default. PERMANENT would mean that the device knows that it is supposed to store permanently but is somewhat free in deciding when to do this. And PERMANENT_IMMEDIATELY should be used only then absolutely necessary. (and GCSes would have to adapt to the new scheme ASAP)

Note that users will have to understand the underlying mechanism for the FLASH-based devices very well!
For instance, doing a PARAM_SET with TEMPORARY followed by a PARAM_SET with PERMANENT will almost surely result in both changes being stored permanently. Even more confusingly, if one does a PARAM_SET with PERMANENT followed by a PARAM_SET with TEMPORARY, it will depend on the time span between the two and the device's internal (and probably unknown) decimation mechanism whether the 2nd parameter change indeed is just temporary or in fact will also become permanent. 

Just my 2 cents. :)

DonLakeFlyer, 2019-05-17:
> So, if you were to ask me, I would undo this move and go back to the "old" spec, where PARAM_SET makes a non-permanent write, and the STORAGE command to make it permanent.

I still see no reason why the ground station should be in control of understanding the intricacies of how a piece of hardware needs to manage it's storage. It has no idea what the requirements for that are. Only the hardware itself does. 

I also see no use case for supporting temporary write to a parameter as an end user feature. 

Put those two things together and the intention of a parameter write is always clear. Save it using whatever mechanism is appropriate for the hardware.

hamishwillee, 2019-05-21:
I'm with @DonLakeFlyer on this one. Might change my mind if we can find a compelling use case for temporary write being something a sender would need :-). Even then though, I think I'd keep the default "auto" as "leave to the device to decide", because that is the reality of all existing implementations.

olliw42, 2019-05-21:
I hear you guys but continue to find the positions unrealistic, for the reasons given. 

>  because that is the reality of all existing implementations.

and that's why things don't work as they should ;).

Anyways, at the end of the day, it's your call, and I understand that well. 

I have a very simple view on this: I would never compromise a component's performance for the sake of some protocol specification. So STorM32 won't work fully with QGC. So be it. Not the biggest disaster on earth. :):)

anyway thx, guys :)

hamishwillee, 2019-05-21:
@olliw42  For your solution we would have to re-write the microservice and autopilot implementations, and we'd still have the unsolved problem that there is a user expectation that parameters are permanent. Fixing it as proposed does not solve the problems for other stakeholders - we would need to do a lot more thinking. 

I wouldn't expect a manufacturer to degrade the device by obeying the protocol - though I might question whether the the device can be fit for purpose if it does not meet the user expectation of permanent parameter storage. 

Anyway, do appreciate your points. Longer term there is discussion replacing this protocol. I will point that discussion to this thread/invite you to next discussions.

olliw42, 2019-05-22:
the point here is that MAVLink once had the perfectly working solution ... and MP still has it, only QGC doesn't ;)

but as said, and I had said it also straightforwardly in the above, the STorM32 is certainly not relevant to you but your stakeholders and whoever is using MAVLink in a professional setting are. And I fully understand this. 

:)

----
just as a side note, it's not correct that my proposed solution would require to re-write autopilots, only QGC would need some little code. The proof in the pudding: MP obviously works with the autopilots without having to rewrite them ;). There would be also no unsolved user expectation problem. The proof in the pudding: MP .... . Also, the STorM32 certainly fully can meet the user expectation of permanent parameter storage. The proof in the pudding: MP .... (I'm frankly not sure if my arguments have been carried on).

hamishwillee, 2019-05-22:
MP doesn't count as a proof because it only works with ArduPilot; If it works with the old approach so long after the spec update is more a sign that the code in that area has not been maintained - not that MP is doing something right.

I guess we'll have to agree to disagree.

olliw42, 2019-05-22:
> MP doesn't count as a proof 

frankly, I think you're starting to twist arguments. but that's OK

> I guess we'll have to agree to disagree.

that's what I'm kind of saying since a couple of posts :)

>  If you can convince

I'm not going to make any effort to convince anyone. :)

I've made sufficient effort. From my perspective the situation is settled. It won't work with QGC, there is still MP, and if MP will make the same deccision, then also MP won't work. I'm not going to work like crazy to implement a piece of spec which compromises the main functioning of the item. I consider it a flaw of the spec. As simple as that. And anyone is welcome to agree or disagree on this :).

I think we really have exchanged and wrapped around all arguments in sufficient depth, and can leave it as it is. As you say, we can agree to disagree. 

:)