mavlink/qgroundcontrol #7434: Improvement: Display Camera Available Space also if only CameraCaptureStatus is available

olliw42, 2019-05-11:
#current behavior
QGC nicely serves the camera micro-service, and displays the available capacity of the camera.

The behavior is related to the two message pairs MAV_CMD_REQUEST_STORAGE_INFORMATION -> STORAGE_INFORMATION and MAV_CMD_REQUEST_CAMERA_CAPTURE_STATUS -> CAMERA_CAPTURE_STATUS.

However, STORAGE_INFORMATION appears to be requested only once at the beginning as part of the camera identification, while CAMERA_CAPTURE_STATUS is requested at the beginning and also repeatedly during video capture. Thus, the actual fresh data for the available capacity is coming from CAMERA_CAPTURE_STATUS.

However, whether or not the available capacity is displayed at all hinges on STORAGE_INFORMATION: If STORAGE_INFORMATION provides a valid value for the storage_total_capacity, then QGC displays the available capacity, and the displayed value is updated according to the new info obtained via CAMERA_CAPTURE_STATUS. All good and nice. But if STORAGE_INFORMATION does not provide valid values, then the available capacity is NOT displayed by QGC, even though it has good values for the available capacity from CAMERA_CAPTURE_STATUS.

#suggested improved behavior
I would suggest to modify the behavior such, that the available capacity is displayed even if the info in STORAGE_INFORMATION is not valid, but CAMERA_CAPTURE_STATUS provides valid data. In short: When CAMERA_CAPTURE_STATUS provides a valid value for the available capacity, it should be displayed.

dogmaphobic, 2019-05-13:
This will require some thought. It was designed with the idea that the camera storage is in place by the time the ground station connects and it won't change during the session. If you are to insert/remove a microSD card for instance, there are no mechanisms to handle this. If `storage_total_capacity` is 0, that means _No Storage_. We will need a way to dynamically be notified of storage changes so the ground station can be in sync with the vehicle.

olliw42, 2019-05-13:
I'm somewhat confused by that response, but that's certainly because I seem to misunderstand the concept behind it.

I wasn't considering a situation where the SD card would be replaced during what you call a session.

My point was much more simple minded. Namely that there can be (and IMHO should be) a useful indication of the available capacity even if there had never been any STORAGE_INFORMATION message, thanks to the fact that the CAMERA_CAPTURE_STATUS has exactly such a field. So my suggestion is to always show  the available capacity, as soon as there is a valid data (non zero and non-NAN) in CAMERA_CAPTURE_STATUS.

not sure if that additional effort at explaining what simple I had in mind did help

dogmaphobic, 2019-05-13:
Well,theoretically, the `STORAGE_INFORMATION` message is mandatory and if it returns 0 for total storage, it means there is none, regardless of what `CAMERA_CAPTURE_STATUS` says. The reason `CAMERA_CAPTURE_STATUS` returns available space is because as you take pictures and/or record video, the storage decreases and that information is not only shown to the user, but also allows the UI to prevent the user from taking pictures and/or recording video when you run out of space. That is, if the original `storage_total_capacity` is 0, the user will never be able to take pictures and/or start recording video and therefore, the GCS won't ever ask for `CAMERA_CAPTURE_STATUS`.

That was the idea until you brought it up. The real problem is that currently, there is no way to tell the GCS of changes to storage availability. That's what needs some thought :)

olliw42, 2019-05-13:
> That is, if the original storage_total_capacity is 0, the user will never be able to take pictures and/or start recording video and therefore, the GCS won't ever ask for CAMERA_CAPTURE_STATUS.

this appears to be not correct. CAMERA_CAPTURE_STATUS is periodically requested even if the total capacity in STORAGE_INFORMATION is set to zero, at least this is what I observe with GCS of two days ago, and I think this also makes sense given the other information available in CAMERA_CAPTURE_STATUS. And since this is so I was thinking that also the available capacity field in CAMERA_CAPTURE_STATUS should be used by the GCS in any case.
I guess I would agree that if it were as you say and CAMERA_CAPTURE_STATUS would not be requested if total capacity is zero (and in fact video taking could not be started), but as said that's not how it appears to work.
maybe that's actually the issue...

olliw42, 2019-05-13:
so to be sure that I'm not talking nonsense I just have double checked this.

I first had storage_total_capacity = 0.0f, with that I could start the video, and I've let it run for ca 16s and stopped it. The result see in the screen shot
[image]

I then had storage_total_capacity = NAN, with that I could start the video, and I've let it run for ca 16s and stopped it. The result see in the screen shot
[image]

so, in both cases I
(1) could start the video
(2) CAMERA_CAPTURE_STATUS  was requested periodically 

just FYI the other values in STORAGE_INFORMATION had been
```
storage_id = 0; //Storage ID (1 for first, 2 for second, etc.)
storage_count = 0; //Number of storage devices
storage_status = 0; //Status of storage (0 not available, 1 unformatted, 2 formatted)
storage_total_capacity = NAN; // 0.0f;
storage_used_capacity = NAN;
storage_available_capacity = NAN;
storage_read_speed = NAN;
storage_write_speed = NAN;
```

hope that helps

olliw42, 2019-05-13:
I would think that if STORAGE_INFORMATION is supposed to be mandatory, and its values to be "meanigfull" (whatever this means needs to be defined), that then the GCS should not show the camera GUI or show it with capacity zero and refuse taking videos/photos (with maybe a warning saying why)
but as it currently is I fail to see a reason why one would not always want to show the available capacity as obtained from CAMERA_CAPTURE_STATUS

dogmaphobic, 2019-05-15:
> then the GCS should not show the camera GUI or show it with capacity zero and refuse taking videos/photos (with maybe a warning saying why)

Yup, that's the bug. QGC is not respecting the "no storage" state. The camera shutter/video record buttons should be disabled and where you normally have the "space available", there should be a warning telling there is no storage. I will deal with that.

olliw42, 2019-05-15:
great that the real reasons for the confusion has then been identified

could it maybe make sense to distinguish three cases:
* storage_total_capacity = NAN: don't show space available, but have camera shutter/video record buttons ENABLED!
* storage_total_capacity = 0.0f: as you say, camera shutter/video record buttons should be disabled and where you normally have the "space available", there should be a warning telling there is no storage
* storage_total_capacity > 0.0f: obvious behavior

The reason for the NAN case is that there are interesting cameras which do not offer SD card storage info, but video on/off, photo on/off modes, and it would be sad if they couldnÄ't be used.

on another matter, when you are at it:
The camera field is first displayed with a disabled "camera trigger" button, when QGC has connected to the pixhawk it enables the "camera trigger" button, when it gets the first info from the camera manager the display changes to these nice buttons but without storage info yet, and a short moment later then the storage info comes in the display again changes to also show the storage info. All fine in principle but a bit "strange".
I would change the behavior to such, that it starts with displaying the disabled "camera trigger" button, and when waits for as long as it needs to conclude what the final display should be, and only when changes it. If that is not appropriate/possible, I at least would ensure that everything remains disabled/deactivated for until QGC has concluded what the final display should be, so that it's clear to the user that she can't yet use it.

dogmaphobic, 2019-05-15:
First of all, what QGC really needs is a real camera control. I stashed that control in there because there was no other place. That control was already there (with the single "camera trigger" button). 

That single "camera trigger" button was the original "dumb camera control". It emits a legacy camera trigger command. If and when a camera supporting the camera API shows up, the control is replaced by one that understands "smart cameras".

With that said, I have no objections to all of your suggestions. The documentation for the camera API will need to be updated to handle the NAN value for storage but that's simple. The sequence of events is also simple (have the UI reflect what is going on based on the discovery). The only thing I'm not sure of is the single "camera trigger" button. That is there for dumb cameras and only replaced if a "smart camera" shows up. It will continue to be there.

While at it, the whole "video streaming control" is also out of place. When you have a "smart camera", it also handles video streaming. All those controls should be together as they are related. It only makes sense to have a video streaming specific control for cases where you have a camera that doesn't support the camera API and only provides video streaming.

This is a typical issue with a GCS that is supposed to support _anything_. There is no easy way to make it behave like a normal GCS where all components are known and predictable.

@hamishwillee For the storage issue above, I will need to update the MAVLink spec here:

https://mavlink.io/en/messages/common.html#STORAGE_INFORMATION

The `total_capacity` field will have this description:

Value|Description
----|----
NAN|Storage not used (camera fully functional without it)
0|Missing storage (no microSD card inserted for example)
&#62; 0|Actual total storage capacity.

dogmaphobic, 2019-05-15:
Now that I look at it, I think it makes more sense to add a proper enum to `status` and add a flag indicating it has no "storage" but the camera should be enabled.

Field Name | Type | Units | Description
-- | -- | -- | --
time_boot_ms | uint32_t | ms | Timestamp (time since system boot).
storage_id | uint8_t |   | Storage ID (1 for first, 2 for second, etc.)
storage_count | uint8_t |   | Number of storage devices
status | uint8_t |   | Status of storage (0 not available, 1 unformatted, 2 formatted)
total_capacity | float | MiB | Total capacity.
used_capacity | float | MiB | Used capacity.
available_capacity | float | MiB | Available storage capacity.
read_speed | float | MiB/s | Read speed.
write_speed | float | MiB/s | Write speed.

dogmaphobic, 2019-05-15:
See here: https://github.com/mavlink/mavlink/pull/1137

olliw42, 2019-05-15:
that's indeed the better approach
excellent

olliw42, 2019-05-15:
>The only thing I'm not sure of is the single "camera trigger" button. That is there for dumb cameras and only replaced if a "smart camera" shows up. It will continue to be there.

I don't mind about this be there, as long as it is disabled until it is either replaced by the when activated/enabled camera GUI display or enabled and not changed anymore. :)
I actually like that button, since if there is no camera stuff, one still can easily trigger a photo. 
Makes sense to me. I just didn't like how the transition from the disabled button to the final display is currently

> it also handles video streaming. 

indeed, this is also still a bit inconsistent. :)
Besides what you say: When the camera capability flag does NOT have a STREAM flag, then the camera menu nevertheless shows a stream id drop down, which is however empty and when selected has no drop downs. I would think that if the STREAM capability is not set that this field should not be displayed.

hamishwillee, 2019-05-16:
OK, some feedback in https://github.com/mavlink/mavlink/pull/1137. Thanks for the heads up.

dogmaphobic, 2019-05-16:
> Besides what you say: When the camera capability flag does NOT have a STREAM flag, then the camera menu nevertheless shows a "Stream Selector" drop down field, which is however empty and when selected has no drop downs. I would think that if the STREAM capability is not set that this field should not be displayed.

Let's put that on a new issue so it doesn't fall through the cracks.

PR #7447 (2019-05-16): Check and respect camera storage status
Following [this conversation](https://github.com/mavlink/qgroundcontrol/issues/7434).

Closes #7434
closed 2019-05-16 by dogmaphobic