mavlink/qgroundcontrol #7436: Survey not centered. Why?

tops4u, 2019-05-12:
When using Survey it will not center Flightpath. If I specify a Spacing of 20m I would expect that it will for sure cover a corridor of 20m. But if the border of the Survey is just 19.5m away then I would miss 9.5m of the Survey Area. 

Illustration:
As you can see on the East-North-East side there is no additional Flightpath. Also the Flighpath is not centered, on the west part it is much to close to the polygon edge. 
[image]

Measurement shows, that if the Aircraft has 20m Coverage, on the east it will miss roghly 9.5m of the Polygon:
[image]

## Expected Behavior
Survey would make sure, that all the Area in the green Polygon is covered. At least I would expect that the Flight Path is centered. 

## Current Behavior
Survey may miss Area close to the Polygon edge. 

## System Information
When posting bug reports, include the following information
- Mac OS Mojave
- QGC Version: 3.5.0 (probably a Daily Build)
- FMUv5 / PX4 1.9rc

DonLakeFlyer, 2019-05-14:
Attach the plan file and I can take a look

tops4u, 2019-05-20:
Ok, I have tried to recreate a similar Case (had not saved the above):

Image from QGC:
[image]

Measurement in the Map:
[image]

Mission File:
[SpacingProblem.plan.zip](https://github.com/mavlink/qgroundcontrol/files/3197563/SpacingProblem.plan.zip)

closed 2026-10-04 by github-actions[bot]