mavlink/qgroundcontrol #7441: Serial Port Transmission can't get full parameter from ArduPilot on QGC 3.5.2

bys1123, 2019-05-14:
## Current Behavior
I got lots of report these days, If people use Radio like P900 or ZigBee that not flashing MAVLink firmware, then they can't get the full parameter list on QGC.
[image]
[image]
[image]

## System Information
When posting bug reports, include the following information
- QGC Version: [3.5.2]
- Flight Controller: [Pixhawk4]
- Autopilot (with version): [Ardupilot Copter 3.6]

## Detailed Description
Provide further details about your issue/bug.

bys1123, 2019-05-14:
Tried change SERIALx_PROTOCAL to MAVLink2, still not work.

DonLakeFlyer, 2019-05-14:
First step is to check you mavlink packet loss rate (Settings/Mavlink). This tends to happen when the communication link is rally bad.

bys1123, 2019-05-15:
[image]
Thanks! It's really have a lot of message loss, but these radio actually very close, and works fine on Mission Planner.

DonLakeFlyer, 2019-05-15:
I'm not sure how/if MP reports that it didn't get all parameters. It might not tell you. Either way I wouldn't be flying with that high of a packet loss.

DonLakeFlyer, 2019-05-15:
Looks like MP basically never gives up trying to get parameters, whereas QGC gives up after 5 failures to get an individual parameter across the wire. So although it may be really slow MP may eventually succeed. I'm not sure if that is a better  approach since at that loss rate something is very wrong. I'd rather have QGC fail in that case and tell you something isn't right instead of going to extreme measures to get all params to download.

DonLakeFlyer, 2019-05-15:
I'm also thinking about adding an additional warning that shows up if the mavlink loss rate goes up beyond a certain amount.

bys1123, 2019-05-16:
Thanks for the information, I think warning for the high mavlink loss is great. I'll trying to find out why they have such amount data loss, it's microhard p900.

bys1123, 2019-05-17:
[image]
Here is another screenshot from other manufacture, work fine for Ardupilot.

bys1123, 2019-05-17:
[image]
But that high loss link is works well on PX4, what's the differents?

bys1123, 2019-05-17:
So, my opinion this is a compatibility issue~

DonLakeFlyer, 2019-05-17:
Are you saying you are now getting a full parameter download even at 9% loss rate?

bys1123, 2019-05-17:
If using PX4 it at 9% loss rate, and not have the much loss after parameters load, finally it will at about 4%. Looks like the parameter list have no problem, maybe have some loss?
But on ArduPilot, the loss rate will much high to 79%.

DonLakeFlyer, 2019-05-17:
Very strange. That would be something for the ArduPilot folks to look at since it is in control of Vehicle->GCS mavlink direction. QGC isn't really in the mix at that point.

dogmaphobic, 2019-05-17:
It could also be an error somewhere in the MAVLink pipeline. If two things use separate instances of MAVLink but the same sysID and compID, the counters will be out of sync. That would cause QGC to think tons of messages are being lost because of the discrepancies in the message sequence.

dogmaphobic, 2019-05-17:
The same could happen if there is something in the middle messing with the messages. Something like a bridge responsible for receiving messages from the autopilot and forwarding them to the GCS. If the payload is forwarded untouched, everything is OK. But if this "bridge" parses _some_ of the messages and repackages them before forwarding, it will have a different sequence number than the original and break the counters as well.

DonLakeFlyer, 2019-05-17:
Hmm, then I wonder if that sysid/compid mapping in there is correct? The sequence numbers are on a mavlink channel basis aren't they? Why does sysid/compid come into play?

DonLakeFlyer, 2019-05-17:
> That would be something for the ArduPilot folks to look at since it is in control of Vehicle->GCS mavlink direction. QGC isn't really in the mix at that point I don't think.

@bys1123 Just FYI: But given discussion this statement may not be correct.

DonLakeFlyer, 2019-05-17:
Also FYI: I just tried with USB connection and got 0 packet loss, so there isn't something inherently wrong with at least a simple ArduPilot connection.

dogmaphobic, 2019-05-17:
> The sequence numbers are on a mavlink channel basis aren't they? Why does sysid/compid come into play?

It's based on the sysID/compID. There is a message_sequence[255][255] array handling it. If component sends messages and _something else_ sends messages using that same sysID/comID combination, it will be using a separate instance of MAVLink and hence, different counters.

bys1123, 2019-05-17:
@DonLakeFlyer Could you try to use FTDI connect to TELEM port?

dogmaphobic, 2019-05-17:
Doe the ArduPilot firmware support UART flow control? That could also be a factor.

bys1123, 2019-05-17:
It's disabled on Pixhawk4, their p900 radio also have no CTS RTS.

bys1123, 2019-05-17:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/Pixhawk4/defaults.parm
```
BRD_SER1_RTSCTS 0
BRD_SER2_RTSCTS 0
```

DonLakeFlyer, 2019-05-17:
> It's based on the sysID/compID.

Hmm. I get what you are saying now but I have a feeling it's clear as mud with respect to mavlink spec as to how that is really supposed to be handled. Specifically at the individual component level with a single system.

dogmaphobic, 2019-05-17:
> Specifically at the individual component level within a single system.

As long as each component only sends messages on its channel, it's all fine. The issue comes up when say, a camera decides to send "autopilot" messages every now and then (setting the compID to 0 for those messages instead of using its own compID). That completely messes up the counters and shows as a high loss rate. There is a company out there doing just that...

DonLakeFlyer, 2019-05-17:
I would say that is you see high loss rate reported and the parameter download fails then the loss rate is likely real, at least to some extent. If you get that high of a loss rate and the parameter download succeeds then the loss rate value is not truly correct.

bys1123, 2019-05-17:
One user tried ESP8266 for ArduPilot, also can't get parameter list, and can't upload waypoint, QGC warning: high legacy link do not support upload.

bys1123, 2019-05-17:
I tried Raspberry Pi as companion for ArduPilot, then connect to QGC, It works fine. Weird~

DonLakeFlyer, 2019-05-17:
> QGC warning: high legacy link do not support upload.

That would mean they setup a UDP link manually and configured it as a High Latency Link which would be incorrect.

bys1123, 2019-05-17:
> If you get that high of a loss rate and the parameter download succeeds then the loss rate value is not truly correct.

I'll check the parameter load duration to confirm this.

bys1123, 2019-05-18:
The loss rate is lower and lower after parameter load, finally at 4%, I think this is true loss rate.

DonLakeFlyer, 2019-05-18:
Then my guess is that the stream rates for telemetry are too high. The link gets flooded out when parameters are streaming down and the rest of the telemetry streams are trying to download as well. Seems like this radio is going to need a custom stream rate setup. You can do that by turning off QGC requesting streams from mavlink settings and then settings the rates yourself using ArduPilot parameters. But I'd also try to talk with other folks who are using ArduPilot plus this radio to see what they are doing.

bys1123, 2019-05-18:
[image]
Oh, thanks, It do have a link rate setting. I'll let them check too.

Professor77, 2019-05-19:
I am new to drones and discovered an issue that made me abort my maiden flight. 

I was trying to set my home position and geofence.

Hitting the upload button in qgc led to the error message "vehicle requesting position out of range"

I checked the packet loss and it was 85%, however the radios are connecting fine for regular telemetry information.

I'm not going to try and fly until I can be sure my drone knows where home is!

bys1123, 2019-05-22:
@Professor77 Thanks for the report. I'll still check on this.

bys1123, 2019-05-23:
Checked the link rate, it's 230400

closed 2026-10-04 by github-actions[bot]